pascalkuthe/OpenVAF #14: Convergence for HBT with Base Current

metroid120, 2022-12-28:
The convergence behavior of HBTs with current biasing at the base is problematic, likely related to the $limit function that 
is missing in Verilog-A models.
 
A minimum netlist is printed below for replication of the issue. Questions that result:
- is the "pnjlim" function working as expected?
- how do commercial Verilog-A compilers handle this?
- Or is it not pnjlim?

```
HICUM2v2.40 Output Test Ic=f(Vc,Ib)

IB 0 B 1u
VC C 0 1.8
VS S 0 0.0

N1 C B 0 S tj npn_full_sh

.control
pre_osdi hicumL2V2p4p0.osdi
altermod @hicumL2V2p40[flsh]=0
op
.endc

.include model-card-examples_osdi.lib

.end
```

where model-card-examples_osdi.lib corresponds to:

```
* Example parameter test set
.model npn_full_sh hicuml2va
*
*Transfer current
+ c10 = 9.074e-030
+ qp0 = 1.008e-013
+ ich = 0
+ hf0 = 40
+ hfe = 10.01
+ hfc = 20.04
+ hjei = 3.382
+ ahjei = 3
+ rhjei = 2
+ hjci = 0.2
*
*Base-Emitter diode currents
+ ibeis = 1.328e-019
+ mbei = 1.027
+ ireis = 1.5e-015
+ mrei = 2
+ ibeps = 1.26e-019
+ mbep = 1.042
+ ireps = 1.8e-015
+ mrep = 1.8
+ mcf = 1
*
*Transit time for excess recombination current at b-c barrier
+ tbhrec = 1e-010
*
*Base-Collector diode currents
+ ibcis = 4.603e-017
+ mbci = 1.15
+ ibcxs = 0
+ mbcx = 1
*
*Base-Emitter tunneling current
+ ibets = 0.02035
+ abet = 24
+ tunode = 1
*
*Base-Collector avalanche current
+ favl = 18.96
+ qavl = 5.092e-014
+ alfav = -0.0024
+ alqav = -0.0006284
+ kavl = 0.0
+ alkav = 0.0
*
*Series resistances
+ rbi0 = 4.444
+ rbx = 2.568
+ fgeo = 0.7409
+ fdqr0 = 0
+ fcrbi = 0
+ fqi = 1
+ re = 1.511
+ rcx = 2.483
*
*Substrate transistor
+ itss = 1.143e-017
+ msf = 1.056
+ iscs = 4.60106e-015
+ msc = 1.018
+ tsf = 0
*
*Intra-device substrate coupling
+ rsu = 500
+ csu = 6.4e-014
*
*Depletion Capacitances
+ cjei0 = 8.869e-015
+ vdei = 0.714
+ zei = 0.2489
+ ajei = 1.65
+ cjep0 = 2.178e-015
+ vdep = 0.8501
+ zep = 0.2632
+ ajep = 1.6
+ cjci0 = 3.58e-015
+ vdci = 0.8201
+ zci = 0.2857
+ vptci = 1.79
+ cjcx0 = 6.299e-015
+ vdcx = 0.8201
+ zcx = 0.2863
+ vptcx = 1.977
+ fbcpar = 0.3
+ fbepar = 1
+ cjs0 = 2.6e-014
+ vds = 0.9997
+ zs = 0.4295
+ vpts = 100
+ cscp0 = 1.4e-014
+ vdsp = 0
+ zsp = 0.35
+ vptsp = 4
*
*Diffusion Capacitances
+ t0 = 2.089e-013
+ dt0h = 8e-014
+ tbvl = 8.25e-014
+ tef0 = 3.271e-013
+ gtfe = 3.548
+ thcs = 5.001e-012
+ ahc = 0.05
+ fthc = 0.7
+ rci0 = 9.523
+ vlim = 0.6999
+ vces = 0.01
+ vpt = 2
+ aick = 1e-3
+ delck = 2
+ tr = 0
+ vcbar = 0.04
+ icbar = 0.01
+ acbar = 1.5
*
*Isolation Capacitances
+ cbepar = 2.609e-014
+ cbcpar = 1.64512e-014
*
*Non-quasi-static Effect
+ flnqs = 0
+ alqf = 0.166667
+ alit = 0.333333
*
*Noise
+ kf = .3e-16
+ af = .75
+ cfbe = -1
+ flcono = 0
+ kfre = 0.0
+ afre = 2.0
*
*Lateral Geometry Scaling (at high current densities)
+ latb = 0.0
+ latl = 0.0
*
*Temperature dependence
+ vgb = 0.91
+ alt0 = 0.004
+ kt0 = 6.588e-005
+ zetaci = 0.58
+ alvs = 0.001
+ alces = -0.2286
+ zetarbi = 0.3002
+ zetarbx = 0.06011
+ zetarcx = -0.02768
+ zetare = -0.9605
+ zetacx = 0
+ vge = 1.17
+ vgc = 1.17
+ vgs = 1.049
+ f1vg = -0.000102377
+ f2vg = 0.00043215
+ zetact = 5
+ zetabet = 4.892
+ alb = 0
+ dvgbe = 0
+ zetahjei = -0.5
+ zetavgbe = 0.7
*
*Self-Heating
+ flsh = 0
+ rth = 1113.4
+ cth = 6.841e-012
+ zetarth = 0
+ alrth = 0.002
*
*Compatibility with V2.1
+ flcomp = 2.3
*
*Circuit simulator specific parameters
+ tnom = 26.85
*+ dt = 0

```

metroid120, 2022-12-28:
I compiled ngspice without the "pnjlim" statements in the built-in HICUM and it shows quite similar convergence to the OSDI model, further strengthening the statement that the convergence is bad due to missing or buggy "pnjlim" in OpenVAF.

pascalkuthe, 2023-01-11:
Closing as the issue here is not with openvaf but simply that models should be using `$limit` with `pnjlim`

closed 2023-01-11 by pascalkuthe