pascalkuthe/OpenVAF #15: Weird Compilation Behavior

metroid120, 2022-12-28:
See attached Ngspice Circuit that uses the attached hicum Verilog-A Code.

If we run the circuit with:

`Vbiei = $limit(V(br_biei), "pnjlim", VT, 1.0);`

in line 1303 Ngspice does not work and says: Error: no such device or model name hicuml2v2p40
If I change the line to 

`Vbiei = $limit(V(br_biei), "pnjlim", VT, 1.0);`

Ngspice does not throw this error.

[issue_compile.zip](https://github.com/pascalkuthe/OpenVAF/files/10314533/issue_compile.zip)

dwarning, 2022-12-28:
Sorry, I see no difference in the modified line. Cut&Paste error?

metroid120, 2022-12-30:
Yes, totally right. I edited my post.

metroid120, 2022-12-30:
After removing the erroneous line with "altermod", it works. Still confused why in one case the model was found and in the other not. Issue closed for the moment.

closed 2022-12-30 by metroid120