pascalkuthe/OpenVAF #17: Missing ground and thermal syntax elements

dwarning, 2023-01-05:
Problems with openvaf version 22.12.0:

ground nodes not accepted:
`    //internal nodes
    electrical gnd; 
    thermal ti;
    ground gnd;
`

Access function for thermal nodes seems not implemented:
`        Texi = Temp(ti);    
`

Example case:

[fbhhbt.zip](https://github.com/pascalkuthe/OpenVAF/files/10353210/fbhhbt.zip)

pascalkuthe, 2023-01-05:
Thanks for the report.
I need to test the `ground` node it should be supported but I haven't tested it much yet 
For the thermal node I believe this is an issue with the model and not OpenVAF.

`Temp` is not build into the compiler but is definied inside `discipline.vams` instead. `Temp` is sued with many other (CMC standard) models and works without issue.
The problem in this model in particular is that there is also a variable named `Temp`. As far as I know all definitions share a namespace in Verilog-A so when OpenVAF looks for `Temp` it (correctly) finds the variable `Temp` instead of the access function `Temp`.

Maybe other compilers treat these functions as built-ins instead of just a normal definition? I am pretty sure the behavior here is consistent with the hieraichal structure described in the Verilog-AMS LRM

dwarning, 2023-01-05:
Yes - the problem with Temp() access function disappears if we rename parameter Temp to DTemp.
ground node problem still exist.

closed 2023-01-12 by bors[bot]

pascalkuthe, 2023-01-12:
The problem with ground nodes has been fixed in master.
As I said I believe that the issue with `Temp` is an issue with the model and handeled correctly by openvaf.

Thank you for testing!

commit 7eb817f7 (2023-01-12): Merge #24

24: fix various small issues r=pascalkuthe a=pascalkuthe

- fix $limit invocation with user-defined functions (fixes #13)
- fix: correctly parse net declarations without disciplines (fixes #17)
- fix panic for model without contributes
- fix: panic if op dependent parameters are added during residual generation (fixes #23)
- perf: avoid producing unnecessary text mappings
- fix regressed syntax errors
- add opt profile for releases
- add build instructions using the docker image
- link website and nightly version in README (addresses #21)
PR #24 (2023-01-12): fix various small issues
- fix $limit invocation with user-defined functions (fixes #13)
- fix: correctly parse net declarations without disciplines (fixes #17)
- fix panic for model without contributes
- fix: panic if op dependent parameters are added during residual generation (fixes #23)
- perf: avoid producing unnecessary text mappings
- fix regressed syntax errors
- add opt profile for releases
- add build instructions using the docker image
- link website and nightly version in README (addresses #21)