pascalkuthe/OpenVAF #35: link error on win10

dwarning, 2023-02-08:
I made first trial under windows 10 64bit. Placed the openvaf.exe in my PATH:

c:\msys64\home\dwarning\spice\Verilog-A\VA-Models\code\vbic\vacode>openvaf.exe --version
openvaf 23.2.0

Compiling the model I got:

c:\msys64\home\dwarning\spice\Verilog-A\VA-Models\code\vbic\vacode>openvaf.exe vbic_1p3.va
Non-UTF-8 output:    Bibliothek \"vbic_1p3.lib\" und Objekt \"vbic_1p3.exp\" werden erstellt.\r\nvbic_1p3.o3 : error LNK2019: Verweis auf nicht aufgel\x94stes externes Symbol \"snprintf\" in Funktion \"cb.2\".\r\nvbic_1p3.osdi : fatal error LNK1120: 1 nicht aufgel\x94ste Externe\r\n
error: linking failed (see linker output for details)
error: failed to compile vbic_1p3.va

Seems a  C stdlib is not into the executable.

pascalkuthe, 2023-02-08:
Could you try this binary: https://openva.fra1.cdn.digitaloceanspaces.com/openvaf_23_2_1_windows_amd64.zip I added a fix that might resolve the problem.

As long as you have the msvc buildtools installed there should be no problem.
From the output it does look like those are installed  so this is a bug

dwarning, 2023-02-09:
Sorry, I see you takes care. Got no email!
Anyway: The new binary didn't work. It shows more link error as the old one. E.g. memcpy and others.

dwarning, 2023-02-10:
Which version of MSVC buildtools you meant? I have VS2022 installed but as I said also in the VS cmdline environment openvaf can't link correct and breaks.
Perhaps commit 357357 is the root cause for the problem. Could you please revert and generate an new executable.

pascalkuthe, 2023-02-11:
I fixed in 458a0ad223a8214a431a4852995c103c76c11fcf. Windows is really a pain to work with, especially because I can't test on it.

You can download the nightly build from the link in the readme for now. I will do a bugfix release later this weekend

closed 2023-02-11 by pascalkuthe

dwarning, 2023-02-11:
Sorry, didn't work:
c:\msys64\home\dwarning\spice\Verilog-A\VA-Models\code\osdilibs>openvaf.exe -D__NGSPICE_COMPACT_MODELING__ -o bsimbulk107.osdi ../bsimbulk/vacode/bsimbulk107.va
Non-UTF-8 output:    Bibliothek \"bsimbulk107.lib\" und Objekt \"bsimbulk107.exp\" werden erstellt.\r\nbsimbulk107.o2 : error LNK2019: Verweis auf nicht aufgel\x94stes externes Symbol \"memcpy\" in Funktion \"concat\".\r\nbsimbulk107.o2 : error LNK2001: Nicht aufgel\x94stes externes Symbol \"_fltused\".\r\nbsimbulk107.o3 : error LNK2001: Nicht aufgel\x94stes externes Symbol \"_fltused\".\r\nbsimbulk107.o4 : error LNK2001: Nicht aufgel\x94stes externes Symbol \"_fltused\".\r\nbsimbulk107.o : error LNK2001: Nicht aufgel\x94stes externes Symbol \"_fltused\".\r\nbsimbulk107.o4 : error LNK2019: Verweis auf nicht aufgel\x94stes externes Symbol \"__chkstk\" in Funktion \"eval_0\".\r\nLINK : error LNK2001: Nicht aufgel\x94stes externes Symbol \"_DllMainCRTStartup\".\r\nbsimbulk107.osdi : fatal error LNK1120: 4 nicht aufgel\x94ste Externe\r\n
error: linking failed (see linker output for details)
error: failed to compile ../bsimbulk/vacode/bsimbulk107.va

dwarning, 2023-02-11:
Solved with you latest build from 20:47,
Thanks

metroid120, 2023-02-11:
Also thanks from my side.