pascalkuthe/OpenVAF #37: $limit(access_function_reference) fails

dwarning, 2023-02-09:
This fails with latest github master:
            Vbe      =  $limit(V(b_be));

this not:
            Vbe      =  $limit(V(b_be), "pnjlim", $vt, tVCrit);

pascalkuthe, 2023-02-09:
This is on purpose, I am not sure if it's documented yet but this can not work and again a case of an impractical aspect of the language standard. In the example above the compiler is supposed to choose a limit function automatically.

In my opinion this not practical. By what criterion would an appropriate limit function be determined. I can't think of a way to determine `vcrit` be? Compact models are usually to complex and almost never contain ideal PN junction which an automatic algopirthm could recognize.

This is a similar issue as with `limexp` where the compiler is supposed to figure things automatically (which is not practical and hence all simulators I know of just liberalize large values).

The only solution that I see here would be to print an warning instead of an error and treat this the same as `V(b_be)`

dwarning, 2023-02-10:
Yes, in my understanding this kind of $limit() call did only inform the compiler that a critical situation happens. It left open what then happens with compiler/simulator.
I agree to print a warning and gives same value back as a result. 
Thanks.