pascalkuthe/OpenVAF #38: Hisim Compilation

metroid120, 2023-02-10:
This issue is related to this [ticket](https://sourceforge.net/p/ngspice/support-requests/62/?limit=25).

When compiling HisimHV 2.5, available [here](https://www.hisim.hiroshima-u.ac.jp/cgi/HiSIM_HV/public_release.cgi), 
I see the following behavior.

**On Linux:**
The model compiles after removing a function 
`//* "smoothUpper" using a power function`
in file  HSMHV_eval_rdrift.inc. 

What bugs me: OpenVAF gives a warning if this function is not removed, but still does not produce the osdi file. 
If this is really a bug, it should be an error. 
After removing the "smoothUpper" Macro, HISIM works for me.

**On Windows:**
The compilation aborts and I see:

`
hisimhv.o3 : error LNK2019: unresolved external symbol snprintf referenced in function cb.2                             hisimhv.o4 : error LNK2001: unresolved external symbol snprintf                                                         hisimhv.osdi : fatal error LNK1120: 1 unresolved externals                                                                                                                                                                                      error: linking failed (see linker output for details)                                                                   error: failed to compile hisimhv.va  `

Attached is a netlist.
[hism_netlist.cir.txt](https://github.com/pascalkuthe/OpenVAF/files/10710508/hism_netlist.cir.txt)

pascalkuthe, 2023-02-10:
On Linux you are likely using an old openvaf version. The bug on windows is duplicate of #35

metroid120, 2023-02-11:
Yes, on windows I was using the old version. So I think we can close this issue.

pascalkuthe, 2023-02-11:
Closing as duplicate of #35

closed 2023-02-11 by pascalkuthe