pascalkuthe/OpenVAF #39: $display() compilation error

bertgoz, 2023-02-14:
Using openVAF 23_2_0 and MSVC (with Desktop development C++ package) I obtain the following compilation error:

> error LNK2019: unresolved external symbol snprintf referenced in function cb.2

dwarning, 2023-02-14:
It is similar to closed issue #35. Please try the nightly build from Readme in front.

bertgoz, 2023-02-14:
Thanks, it now compiles but the nightly release is not compatible with ngspice-39

dwarning, 2023-02-14:
Why?

bertgoz, 2023-02-14:
I get "internal error -- segmentation violation" when running a circuit containing the compiled module with $display() in ngspice-39 Windows.

Without $display() the circuit runs fine (both in the openVAF official release and the nightly)

pascalkuthe, 2023-02-15:
Thank your for reporting this bug. The same problem was already reported on the ngspice user forum.

The segmentation fault is windows specific and seems to be a problem with the way the C standard library works on windows. Sadly linking to the standard library (as a compiler) is a lot more challenging on windows (whereas the something is almost trivial on macos/linux).

I am actively working on the problem but its a bit challenging to solve.

closed 2023-04-06 by bors[bot]

pascalkuthe, 2023-04-06:
thank you so much for digging into this. This does indeed seem to fix the problem. I was totally going in the wrong direction by looking at the standard library. There is simply so much complexity involved with targeting windows that I missed the obvious. The fact that getting any kind of debugging setup on windows in VM made this a pain to debug and I just put it off

EternalXY, 2023-04-06:
:thumbsup:

commit 6ab53652 (2023-04-06): Merge #56

56: export osdi_log symbol on windows r=pascalkuthe a=pascalkuthe

Fixes #39
PR #56 (2023-04-06): export osdi_log symbol on windows
Fixes #39