pascalkuthe/OpenVAF #40: port directions should be optional

KenKundert, 2023-02-23:
Port directions are often not specified for Verilog-A models because the simulator does not use the port direction.  It is primarily needed in Verilog-AMS when inserting connect modules.  By insisting on port directions you are making OpenVAF incompatible with standard-complient models and forcing user to add extra clutter to their models.

pascalkuthe, 2023-02-23:
Hello Ken, thank you for reaching out and giving detailed feedback!

OpenVAF aims to implement the Verilog-A subset defined in Annex C of the  [Accellera Verilog-AMS Language Reference Manual (2.4.0)](https://www.accellera.org/images/downloads/standards/v-ams/VAMS-LRM-2-4.pdf).  As far as I am aware Verilog-AMS itself does not allow port declarations without associated directions (input/output/inout).  Excerpt from section 6.2 of the LRM:

>  The optional list of ports or port declarations shall specify an ordered list of the ports of the module. The order used in defining the list of parameters in the module_parameter_port_list and in the list of ports can be significant when instantiating the module (see 6.2.2). The identifiers in this list shall be declared in input, output, or inout declaration statements within the module definition. Ports declared in the list of port declarations shall not be redeclared within the body of the module.

We interpreted this to mean that the list of ports declarations is optional, but if a module has ports, they must all require directions. I was not able to find any exception to this rule in the language standard.  What part of the standard are you referring to here exactly?

KenKundert, 2023-02-23:
I am not pulling from the standard.  Rather I am pulling from long experience with the Cadence simulators.  I don't know if other simulators are the same, but Spectre and AMS-Designer do not require port direction declarations for ports.  And since in Verilog-A they are no-ops, many existing hand written models do not have port directions.  I usually do not specify port directions for Verilog-A models because they are largely clutter, and can be confusing.  The concept of input and output is only meaningful on electrical ports as statements of intent.  Any electrical port can act as input or output.  For example, consider a supply.  The supply port acts like an input for voltage but and output for current.

I can see your reasoning.  It is sound.  But I prefer the Cadence approach of making it optional.

pascalkuthe, 2023-02-24:
I personally agree with that the concept of port direction doesn't server much purpose in analog models. Nonetheless one of the design philosophies of OpenVAF is a focus on compatibility with the LRM where possible. I would rather see the standard improve in that regard rather than letting tools diverge from the standard. This leads to an -essentially- implementation defined languages which makes reusing models across different compilers hard.

That being said I do definitely see your position in this case and being compatible with cadence is probably something we want to aim for (due to its market status). Ideally models shouldn't need to be changed to work with OpenVAF.

I added a lint system to OpenVAF for situations like this: Just like many C compilers (or the rust compiler) OpenVAF allows the user to compile the severity of some diagnostics (called lints). 

I will change the diagnostic for missing port directions from a hard error to a lint, which defaults to an error.
The user can then change the error to a warning by calling OpenVAF as `openvaf -W ports_without_direction` or totally ignore the warning with `openvaf -A ports_without_direction` (longform of these arguments are `--warn` and `--allow`). 

When the error occurs OpenVAF will also provide a hint to the user that these options exist.
I think this presents an good compromise to you between adhering to the LRM and supporting code

KenKundert, 2023-02-24:
There is a weird dissidence here.  The LRM requires a declaration that makes no sense for most device models, which is your primary focus.  Are the ports of a resistor inputs or outputs?  I know most people go with inout, but in Verilog inout means some times input and other times outputs, so it is not really appropriate either.

I think the solution you propose is good, or you could just ignore this issue.  It is very little trouble adding the direction declarations.  I raised the issue because I was surprised by the error.  It was only after I got your first response that I dug in an confirmed that the LRM requires direction.

pascalkuthe, 2023-02-24:
I implemented the solution I described above in https://github.com/pascalkuthe/OpenVAF/pull/47.

> There is a weird dissidence here. The LRM requires a declaration that makes no sense for most device models, which is your primary focus. Are the ports of a resistor inputs or outputs? I know most [...]

While we are definitely focused on device models right now we do hope to support larger parts of the Verilog-A standard in the future. From my understanding of the LRM `input` and `output` don't actually control the flow direction in anyway. Instead the direction here is entirely semantic:

* `input` means that a branch contend to a port must not be used on the lefthand of a contribute statement
* `output` means that a branch connected to a port must not be used with a probe function (so `x = V(a,b)` is illegal but `V(a,b) <+ x` is ok). 

Relevant excerpts from the LRM:

Section 5.6.1:

> Contributions cannot be made to an analog port declared with an input direction, but an output port can be
accessed

I was actually not able to find a reference for the first rule again although I am sure I found it in the past. However the following example for a simple amplifier from the LRM (section 5.6.1.1) follows the same rule:

An ideal amp without impedances uses `input`/`output` as `in` is only probed and `out` is only driven.
``` 
module amp(out, in);
	input in;
	output out;
	electrical out, in;
	parameter real Gain = 1;

	analog
		V(out) <+ Gain*V(in);
endmodule
```

As soon as impedance/resistances are added the ports need to becomes `inout` as resistors always require both a probe and a contribution (of the same branch):

```
module amp(out, in);
	inout out, in;
	electrical out, in;
	parameter real Gain = 1, Rin = 1, Cin = 1, Rout = 1, Lout = 1;
	analog begin
		// gain of amplifier
		V(out) <+ Gain*V(in);
		// model input admittance
		I(in) <+ V(in)/Rin;
		I(in) <+ Cin*ddt(V(in));
		// model output impedance
		V(out) <+ Rout*I(out);
		V(out) <+ Lout*ddt(I(out));
	end
endmodule
```
So the port direction is really just about providing assurances to a user of a module what the module can/can't do. I personally don't really see a practical usecase for this in Verilog-A (without mixed signal simulations). Especially for a device model that aims to capture physical behavior there is no such thing as port without any impedance. But the definition does atleast appear consistent/well defined.

These rules are currently implemented in OpenVAF. However it seems that other commercial simulators do also not enforce port directions if present. Therefore I will likely turn these into lints in the future so they can be disabled.

KenKundert, 2023-02-24:
> Section 5.6.1:
>
>    Contributions cannot be made to an analog port declared with an input direction, but an output port can be accessed

That is a mistake in the LRM. Ignore it.

>  - input means that a branch connected to a port must not be used on the left hand of a contribution statement
>  - output means that a branch connected to a port must not be used with a probe function (so x = V(a,b) is illegal but V(a,b) <+ x is ok).

If those are in the LRM they are mistakes too.  Ignore them too.

The direction specifier was inherited from Verilog, it has no significance in an circuit simulation.  It was retained in Verilog-A primarily for purposes of documentation.  It allows the user to state their intent about how the ports are to be used.

Fundamentally electrical ports have two signals associated with them, voltage and current.  And if one is an input, the other is generally an output.  Think of a supply port like `vdd`.  The` V(vdd)` is an input, `I(vdd)` is an output.  So one cannot read too much into the direction specifiers.  They are simply declarations of intent.

The direction specifiers only become significant in Verilog-AMS when they are used to determine which type of connect module to insert.  Generally, _inout_ connect modules are more complex and much more troublesome than the connect modules used on _input_ or _output_ pins.  Thus, the use of the _inout_ port direction should be minimized as it generally causes trouble when it affects connect module selection.

The following example demonstrates fallacy of the statements you found in the LRM.  If someone got carried away in how they modeled a logic buffer in Verilog-A, they might end up with something like this:
```
module buffer(out, in);
output out;
input in;
electrical out, in;
parameter real l_out=1n from [0:inf);
parameter real c_in=1p from [0:inf);
parameter real vdd=1;
parameter real thresh=vdd/2 from (0:Vdd);
parameter real tt=10n from (0:inf);
integer in_val;

analog begin
    // model buffer
    @(cross(V(int) - thresh))
        ;
    if (V(in) > thresh)
        in_val = 1;
    else
        in_val = 0;
    V(out) <+ vdd*transition(in_val, 0, tt);

    // model parasitics
    I(in) <+ c_in*ddt(V(in));
    V(out) <+ l_out*ddt(I(out));
end
endmodule
```
Without the parasitics then `in` is clearly an input and `out` is clearly an output. With the LRM as written, the addition of the parasitics forces `in` and `out` to be declared as `inout` even if `c_in` and `l_out` were 0. This results in the wrong connect modules being inserted.  That is clearly ridiculous.

And the restriction on outputs that `x = V(a,b)` is illegal but `V(a,b) <+ x` is just silly.  They are logically equivalent, how can one be allowed and the other be forbidden.

Those comments should be ignored, and eventually removed from the LRM.

pascalkuthe, 2023-02-25:
> Those comments should be ignored, and eventually removed from the LRM.

Thank you for clarifying this. I try to follow the LRM as close possible as that is the only official reference for what the correct behavior should be. I sometimes suspect oversights in the LRM but when complying with the LRM only requires little effort (as in this case) I do so regardless as I have no way to be sure. I will remove these two errors from OpenVAF then.

KenKundert, 2023-02-25:
I am composing a letter to the LRM committee pointing out the error.  I'll add you to the cc list so you can see their response.

KenKundert, 2023-02-25:
The message bounced.  The committee seems disbanded.

pascalkuthe, 2023-02-25:
> The message bounced. The committee seems disbanded.

If I rember right, Geoffrey Coram (from analog devices) briefly mentioned that there was renewed activity at the VAMS comitte (after somebody else pointed out that the committee was disbanded) during the MOSAK last December. Perhaps there is a new address now.  Maybe its possible to reach out to him regarding this issue?

commit 88cadb03 (2023-02-25): Merge #48

48: do not check port direction at contribute/probe r=pascalkuthe a=pascalkuthe

Closes #40, see discussion there for details
commit 9575d321 (2023-02-25): Merge #48

48: do not check port direction at contribute/probe r=pascalkuthe a=pascalkuthe

Closes #40, see discussion there for details
PR #48 (2023-02-25): do not check port direction at contribute/probe
Closes #40, see discussion there for details
closed 2023-02-25 by bors[bot]