pascalkuthe/OpenVAF #45: the parser seems stuck

KenKundert, 2023-02-23:
I have many Verilog-A models that currently cause OpenVA to crash.  Perhaps this is because they are using features of the language you do not yet support.  I largely write functional models rather than device models.  Anyway, if you want to try it, here is one such model:
```
`include "disciplines.vams"

module inv (out, in);
output out;
input in;
electrical out, in;
parameter real vdd=1.8;
parameter real tt=1u;

analog begin
    @(cross(V(in) - vdd/2))
        ;
    V(out) <+ vdd*transition(V(in)<vdd/2, 0, tt);
end
endmodule
```
The problem seems to be with the event statement.  If I comment it out I do get an error message indicating that transition is not yet supported.

Do you want my other examples?  If so, how should I get them to you?

pascalkuthe, 2023-02-24:
Thank you for reporting this bug!

OpenVAF doesn't support most event statements at the moment I am afraid. These crashes likely result from a bug in the parser for the event code. This should not be happening (instead a helpful error should appear). I will look into a fix. The parsing code for events is not well tested at the moment (as events are not supported). Having more examples to ensure that the parser doesn't get stuck would be great!

Could you send me these other examples as an email? My mail address is pascal.kuthe@semimod.de.

pascalkuthe, 2023-02-24:
This actually turned out to be a duplicate of #32 and is already fixed on master

closed 2023-02-24 by pascalkuthe