pascalkuthe/OpenVAF #50: Recording output variables during transient simulation

bertgoz, 2023-02-27:
Using ngspice-39 I am trying to save OPVARs during a transient simulation. When saving a vector of the right dimensions (same as simulation steps) is created but all its values are 0 except the first one.

Is it possible to save the value of OPVARs during all the steps of a transient simulation?

pascalkuthe, 2023-02-28:
Thanks for reaching out. 

This is not an issue with OpenVAF itself. The simulator can easily control when operating point variables are calculated by setting a simple variable. Currently our integration code in ngspice only enables operating point calculation during AC/DC analysis but not during tran (the first `tran` operating point counts as a DC simulation). Fixing this in ngspice would be quite trivial and only require:

Replacing

``` C
  if (is_init_smsig || is_sweep) {
    sim_info.flags |= CALC_OP;
  }
```

with

``` C 
    sim_info.flags |= CALC_OP;
```

in `osdiload.c`. I am quite busy at the moment, could you submit an appropriate patch to ngspice @metroid120?

I am closing this issue here as no changes to OpenVAF itself are necessary, only ngspice itself needs to be adjusted.

closed 2023-02-28 by pascalkuthe

metroid120, 2023-02-28:
Yes I will have a look, thanks for the hint.

metroid120, 2023-03-01:
I pushed a fix to branch "fix_calcop_transient" in Ngspice, it is working then.