pascalkuthe/OpenVAF #51: Question of "Linker not found".

SeaFalling, 2023-03-02:
Hi, I'm using OpenVAF to handle the BSIM-CMG, but when I using OpenVAF 23_2_0 in Windows, there are two errors:

error: linker not found: program not found
error: failed to compile bsimcmg.va

Docs of OpenVAF refered that "On Windows the MSVC build tools must be downloaded and installed [from the microsoft website](https://visualstudio.microsoft.com/downloads/?q=build+tools#build-tools-for-visual-studio-2022)." But after I installed the BuildTools, the errors are still here.
But when I using the OpenVAF 22, the ".va" file can be compiled. I wander if there are anything I need to install, thanks.

metroid120, 2023-03-02:
I am not an expert for Windows, but it is very likely caused by MSVC build tools not being installed as they should be. 

With OpenVAF 23 we removed LLD integration that made linking OpenVAF for windows super difficult, so it is expected that you will only encounter this problem with OpenVAF Version > 23. 

Did you restart your computer? There is a lot of stuff on the internet regarding this error.

SeaFalling, 2023-03-04:
Thanks for your reply. I found the Windows 11 SDK is needed before using the build tools. I can use OpenVAF 23 after installing it in visual studio build tool.

closed 2023-03-06 by metroid120