pascalkuthe/OpenVAF #52: panic and implicite

dwarning, 2023-03-06:
I know implicit it is not recommended, but at least no panic for accessing port current:

        Id = I(anode);

in the attached diode example.

thread '<unnamed>' panicked at 'no entry found for key', openvaf/osdi/src/eval.rs:147:53
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace

[diode_implicit.va.txt](https://github.com/pascalkuthe/OpenVAF/files/10898825/diode_implicit.va.txt)

pascalkuthe, 2023-03-06:
this is definitely a bug and should not cause a panic, thanks for reporting.

That being said this model is incorrect and can never work correctly. `I(anode)` is equivalent to `I(anode, ground)` and threfore creates a new branch to ground.  There are two ways to implement this properly:

* Use `I(<anode>)` (return current flowing trough the `anode` terminal): very much not recommended as it requires two additional nodes. Currently `I(<anode>)` also returns `0` in OpenVAF, the reason is that there are many cases where `I(<X>)` is just used for calculating operating point variables and in that case most simulators offer a way to calculate lead currents without requireing two additional nodes. I want to ensure I can take advantage of that before enabeling this feature in OpenVAF to avoid unnecessarily slowing down compact models
* The recommended way: Use `I(anode, cathode)` still requires one more node than an explicit voltage only formulation but should work well for the most part

I will look into fixing the panic

Marvelousmay, 2023-03-11:
I'm the original questioner, and I'm trying to convert a model built using VHDL-AMS into Veriloga.The limiter part of it encountered the problem that the undefined current could not be used, how to solve such problems?
Below is my VHDL-AMS model：
`-------------------------------------------------------------------------------
-- Entity     : limiter
-------------------------------------------------------------------------------
-- Description: Current limiting, agd ground, P1 input, P2 output
-------------------------------------------------------------------------------
library IEEE;
use IEEE.electrical_systems.all;

entity my_limit is
  generic (
    MAX_value    :     real := 1.0);        -- Gain  
    
  port (terminal agd : electrical;
        terminal p1, p2 : electrical);
end entity my_limit;

architecture ideal of my_limit is

  quantity v across p1 to p2;
  quantity i through p1 to agd;
  quantity iout through agd to p2;

  constant MIN_value : real := -MAX_value;
begin		
  
  v == 0.0;
 
  if i'above(MAX_value) use
     iout == MAX_value;
  elsif not i'above(MIN_value) use
     iout == MIN_value;
  else
     iout == i;
  end use ;

  break on i'above(MAX_value),i'above(MIN_value);

end architecture ideal;

-------------------------------------------------------------------------------
-- Entity     : LDO
-------------------------------------------------------------------------------
-- Description: This is derived from the LDO simple schematic
-------------------------------------------------------------------------------

library IEEE,SPICE2VHD,EDULIB;

use IEEE.ELECTRICAL_SYSTEMS.all;

use SPICE2VHD.SPICE_PARAMETERS.all;

use IEEE.MATH_REAL.all;

entity LDO is

	generic (
			DCref: voltage := 2.5; -- Internal reference voltage
			
			SINE: voltage := 0.0; --Disturbance voltage peaks
			SINE_F : real := 1000.0; --Perturbation frequency

			VOUT_ref : real := 5.0;--Output voltage value ֵ			
			Limit_current : real :=10.0; --Maximum output current
	
			Rin   : resistance := 1.0e6;    -- Input resistance [Ohms]
            Rout  : resistance := 100.0;    -- Output resistance (Ohms]
            Avol  : real       := 100.0e3;  -- Open loop gain
            F_0dB : real       := 1.0e6     -- Unity Gain Frequency [Hz]	    
	);
	
	port(
		 terminal input : electrical;
		 terminal output : electrical;
		 terminal agnd  : electrical
	);
end ;

architecture behav of LDO is 
    terminal N2,N3,N5,N6,N7: electrical;
     
	constant BJT_PARA	: BJT_DATA	 := SET_BJT_DATA	(
			MODEL => NPN ,ISS=>5.3314e-12, 
			BF=>176.704, NF=>1.2565, VAF=>80.5084, IKF=>0.1789, 
 			ISE=>85.634e-15, NE=>1.4657, BR=>15.0, NR=>1.2, 
 			VAR=>23.75, IKR=>0.6816, ISC=>16.5274e-15, NC=>1.495, 
 			RB=>8.8, IRB=>44.725e-3, RBM=>2.2, RE=>15.0e-3,
 			RC=>0.6, CJE=>22.49e-12, VJE=>1.0, MJE=>0.415, 
 			TF=>0.468e-9, XTF=>0.46, VTF=>5.0579, ITF=>0.1955 ,
			CJC=>14.48e-12, VJC=>0.52, MJC=>0.398, XCJC=>0.61 ,
 			TR=>25.5871e-9,XTB=>1.4187,XTI=>2.56,FC=>0.493
 			); 
 		
 	constant R1 : real := 1000.0;
 	constant R2 : real := R1*(VOUT_ref/DCref-1.0);
 	

begin
		
	VSINE : entity EDULIB.V_SINE(IDEAL)
		generic map(freq => SINE_F,amplitude => SINE)
		port map(N2, input);
		
	NPN: entity SPICE2VHD.BJT(SPICE) 
		generic map(MDATA=>BJT_PARA)
		port map(NC=>N2,NB=>N3,NE=>N7);

	OPAMP: entity EDULIB.OPAMP_3P(BASIC)
		generic map(rin => Rin,rout => Rout,avol => Avol,f_0dB => F_0dB)
		port map(in_pos=>N5,in_neg=>N6,output=>N3);
		
	R_1: entity EDULIB.RESISTOR(IDEAL)
		generic map(res => R1)
		port map(N6,agnd);
		
	R_2: entity EDULIB.RESISTOR(IDEAL)
		generic map(res => R2)
		port map(N7,N6);
		
	V2: entity EDULIB.V_CONSTANT(IDEAL)
		generic map(level => DCref)
		port map(N5,agnd);
		
    LIMITER:entity my_limit(ideal)
    	generic map(Limit_current)
    	port map(agd => agnd,p1 => N7,p2 => output);

end architecture behav;`

PR #67 (2023-05-10): Prepare for release 23.5.0
- Migrate to LLVM 15 and opaque pointers
- fix panic for current probes to trivially open branches (fixes #52, fixes #53, fixes #66)
- warn when using trivially open branches
- add basic logging
- add crash handler
- update build instructions
- bump dependencies
closed 2023-05-10 by pascalkuthe