pascalkuthe/OpenVAF #53: Better error

metroid120, 2023-03-10:
This issue refers to this thread:
https://sourceforge.net/p/ngspice/discussion/133842/thread/a0bf455de8/?limit=25

I guess OpenVAF throws an error when a branch current depends on another current that is not defined. While this is of course incorrect, there should be an error message instead of a crash.

pascalkuthe, 2023-03-10:
Interesting, I thought I had already fixed this. Were you ablenyo reproduce with the latest master?

metroid120, 2023-03-10:
yes, I was.

pascalkuthe, 2023-03-10:
Hmm thanks will take a look probably similar to (or even the same bug as) #52

Marvelousmay, 2023-03-11:
> 

I'm the original questioner, and I'm trying to convert a model built using VHDL-AMS into Veriloga.The limiter part of it encountered the problem that the undefined current could not be used, how to solve such problems?
Below is my VHDL-AMS model：

[ldo.v.txt](https://github.com/pascalkuthe/OpenVAF/files/10947447/ldo.v.txt)

My code to model limiter using Verilog-a is as follows:：

`
`include "constants.vams"
```include "disciplines.vams"
module my_limit(p1,p2,gnd);
   	inout p1,p2,gnd;
    	electrical p1,p2,gnd;
	//ground gnd;

	parameter real g = 0.0 from [0:inf);
	parameter real  MAX_value=1.0 from [0:inf);
	parameter real  MIN_value=-1.0 from (-inf:0];

	branch (p1,gnd) n1;
	branch (gnd,p2) n2;
	branch (p1,p2) n3;

	real I1, I2;
	//integer i3;

    analog begin
	
	V(n3) <+ I(n3)*g;
	
	//I(n1) <+ g*V(n1);

	I1 = I(n1);
	//I2 = I(n2);

	if (I1 > MAX_value) begin
	I2 = MAX_value;
   end
   	else if (I1 < MIN_value) begin
	I2 = MIN_value;
   end
   	else begin
	I2 = I1;
    end 
	I(n2) <+ I(n2)*g + I2;
	end
endmodule`

PR #67 (2023-05-10): Prepare for release 23.5.0
- Migrate to LLVM 15 and opaque pointers
- fix panic for current probes to trivially open branches (fixes #52, fixes #53, fixes #66)
- warn when using trivially open branches
- add basic logging
- add crash handler
- update build instructions
- bump dependencies
closed 2023-05-10 by pascalkuthe