pascalkuthe/OpenVAF #54: Module attributes

dwarning, 2023-03-23:
It seems module attributes are accepted by openVAF:
`(* ngspiceModelGroup="semi", ngspiceLevelNumber="1", ngspiceDeviceName="nigbt", ngspiceTypeVariable="N" *) module nigbt (a, g, c);`
But the question is, where they are will landing and how can I use them in simulator?

pascalkuthe, 2023-03-23:
Attributes are an extension mechanism for Verilog-A. They are intended to allow adding extra information not covered by the standard.  However, there is basically no definition of what to do with them except for a couple of standard attributes so OpenVAF ignores all attributes it doesn't know. This mostly matches commercial simulators as far as I know.

Any kind of attribute like this need to be added to the OSDI standard to show up in the simulator. I think we could add a generic `group` attribute (OpenVAF already supports something like that for parameters I think) but the rest doesn't make much sense to me. The way OSDI is integrated into ngspice it's possible to use any other type variable than `N` (and level is not supported at all) while the device name is just the module name.

dwarning, 2023-03-24:
Yes - that is what I meant. Can you provide a structure with module attributes in the ngspice/osdi interface and fill them by OpenVAF. 
Please mark it as feature request.
Thank you.