pascalkuthe/OpenVAF #55: Verilog-A idt operator supported?

EternalXY, 2023-04-06:
Trying a simple simulation with OpenVAF and ngspice, I noticed that the result of the integration operator never changes, is it not supported or am I doing something wrong?
The integrated value changes when simulating with Cadence Spectre, but remains stuck at 0 in OpenVAF.

Verilog code:

```
`include "disciplines.vams"

module testidt(outreal, gnd);
output outreal;
inout gnd;
electrical outreal;
electrical gnd;
real phase;

analog begin
        phase = idt (1, 0);
        V(outreal) <+ phase;
        $strobe("%f %f\n", $abstime, phase);
end
endmodule
```

ngspice circuit:

```
* OSDI test
.control
pre_osdi testidt.osdi
.endcontrol
.model mytest testidt

N1 int 0 mytest
.tran 1s 10s
.PRINT TRAN V(int)
```

pascalkuthe, 2023-04-06:
the idt operator is supported although it's not been tested very well (compact models usually only use ddt). I fixed a bug that caused the miscompilation here. Once #57 is merged (and the master pipeline has run so you might need to wait a while) the nightly build should update and the example you provided produces the expected result. Thank you for providing a minimal reproduction case that helped with nailing down the source of the bug

closed 2023-04-06 by bors[bot]

dwarning, 2023-12-03:
I am not sure but it seems that the original problem files from @EternalXY not run as expected with the actual beta releases.

commit 95990b16 (2023-04-06): Merge #57

57: fix idt operator r=pascalkuthe a=pascalkuthe

Fixes #55
PR #57 (2023-04-06): fix idt operator
Fixes #55