pascalkuthe/OpenVAF #65: openvaf model not simulating correctly

medwatt, 2023-05-02:
I am testing a memristor model from [here](https://knowm.org/well-posed-memristor-modeling-with-xyce-and-verilog-a/) called "Joglekar Resistance Switch Model in Verilog-A". After some minor modifications, it was able to compile with `openvaf`. The result produced is not what I was expecting.

This is the current `I(V1)`  plot from ngspice.

[image]

Of course, there's the possibility that I introduced a mistake when I modified the file. So, I simulated the same verilog-a model in spectre. The plot of the current `I(V1)`  is shown below.

[image]

The modified verilog-a file is given below.

```systemverilog
`include "disciplines.vams"

`define X_BORDER_BUMP 10e-18

module joglekar_memristor (p,n);

    inout p,n;
    electrical p,n;

    parameter real Roff=16000 from (0:inf);
    parameter real Ron=100 from (0:inf);
    parameter real Rinit=11000 from (0:inf);
    parameter real D=10n from (0:inf);
    parameter real uv=10e-15 from (0:inf);
    parameter real p_coeff=1.0 from (0:inf);

    // local variables that should persist over time steps
    real w_last;
    real time_last;

    // local variables that hold temporary values
    real G;
    real window_function;
    real w;
    real dw;
    real R;
    real direction;
    real mem_current;
    real abstime;
    real time_delta;

    analog begin

    @(initial_step)
    begin
        w_last = ((Roff - Rinit) / (Roff - Ron)) * D;
        time_last = 0;
    end

    // calculate conductance
    G = 1 / (Ron * w_last / D + Roff * (1 - w_last / D));
    mem_current = G * V(p,n);

    direction = 0;
    if (mem_current > 0) begin
        if(w_last<=0) begin
            direction=1;
        end
    end
    else
    begin
        if(w_last>=D) begin
            direction=-1;
        end
    end

    abstime = $abstime;
[... 532 lines cut ...]
	-3.506858118480557e-01
	-2.312509760211469e-05

 103	9.628000000000007e-01
	-2.779344913198829e-01
	-1.938801116901117e-05

 104	9.728000000000007e-01
	-2.040862903306625e-01
	-1.514600586823299e-05

 105	9.828000000000007e-01
	-1.294326539916630e-01
	-1.028423975906868e-05

 106	9.928000000000007e-01
	-5.426820609951436e-02
	-4.649617059142608e-06

 107	1.000000000000000e+00
	-2.939152317953648e-16
	-2.671956652685134e-20
```

I did these tests on: `openvaf 23.2.0` and `ngspice 40`

pascalkuthe, 2023-05-03:
The problem is that OpenVAF does currently not support hidden state.

>  // local variables that should persist over time steps
>    real w_last;
>   real time_last;

OpenVAF will not persist variables over timestamps as this makes it impossible to optimize a lot of compact models correctly. I want to offer an attribute to mark a variable has hidden (and a cli setting that does so for all variables automatically which can not be proven to be statefree). This is not quite compliant with the Verilog-A standard but sadly a few compact models are written such that it is NP-hard to determine whether some code is operating point dependent because variables are not initialized properly.

I hope to implement this in the not-so-distant future. There is currently a push from the CMC to make sure compact models do initialize all variables correctly so hopefully one day OpenVAF can just assume that all uninitialized variables are hidden state

medwatt, 2023-05-03:
Thanks for the clarification. Is there a workaround for this issue at the amount (rewriting the code in some way to avoid the need for variables that persist between timestamps)? A lot of memristor models depend on persistent variables.

I have some minor suggestions if you don't mind:
1. The user should be notified that the code he's running requires persistent variables and the simulator does not support that at the moment.
2. I noticed that the simulator failed to warn me when I had a variable defined as one of the keywords in verilog-a. For example, the original code had user-defined variables `time` and `current`. The simulator complained about `time`, but not about `current`. spectre complained about both.

pascalkuthe, 2023-05-04:
> The user should be notified that the code he's running requires persistent variables and the simulator does not support that at the moment.

This is actually a bit hard to detect. To the compiler these non-ideal compact models I described earlier are indistinguishable from the cases I described before. So automatically adding a warning (which I had implemented before) caused many of warnings while compiling CMC models. While generating these warnings is more correct the compact model uses is the largest industry use case and I don't want to degrade the experience here. On the other hand such warnings might push model developers to finally fix those issues.

> I noticed that the simulator failed to warn me when I had a variable defined as one of the keywords in verilog-a. For example, the original code had user-defined variables time and current. The simulator complained about time, but not about current. spectre complained about both.

OpenVAF only treats the identifiers listed in `Table B.1—Reserved keywords` of the Verilog-AMS LRM. `current` is not listed there. `current` is only used as a nature name in the Verilog-A standard library. OpenVAF fully supports defining custom disciplines/natures and the standard library are just normal files processed as normal by the compiler. So `current` receives no special casing in the compiler whatsoever. Paramters are defined at the module scoping level so they are allowed to shadow identifier from a parent scope (in this case natures which are defined at the root scope).

Most established Verilog-A compilers don't stick to the LRM as closely as OpenVAF does so I suspect spectre just treats `current` as a compiler builtin.