pascalkuthe/OpenVAF #66: Compiler paniking when access the current through a port

medwatt, 2023-05-04:
Consider the line `Itb = I(mem_plus, mem_minus);` in the code below. Compiling this file causes the error below. It compiled in spectre just fine. Using  `Itb = I(<mem_plus>);`  resolves the issue. Is this a bug or is non-standard code?

```
thread '<unnamed>' panicked at 'no entry found for key', openvaf/osdi/src/eval.rs:147:53
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```

```systemverilog
`include "constants.vams"
`include "disciplines.vams"

module mem_ddt (mem_plus, mem_minus, gap_value, ddt_value);

    input electrical mem_plus, mem_minus, gap_value;
    output electrical ddt_value;

    parameter real      kb        = `P_K;
    parameter real      q         = `P_Q;
    parameter real      g0        = 0.25e-9  from (0:2e-9);
    parameter real      V0        = 0.25     from (0:10);
    parameter real      Vel0      = 10       from (0:20);
    parameter real      I0        = 1000e-6  from (0:1e-2);
    parameter real      beta      = 0.8      from (0:inf);
    parameter real      gamma0    = 16       from (0:inf);
    parameter real      T_crit    = 450      from (390: 460);
    parameter real      T_smth    = 500      from (400: 600);
    parameter real      Ea        = 0.6      from (0:1);
    parameter real      a0        = 0.25e-9  from (0:inf);
    parameter real      T_ini     = 273 + 25 from (0:inf);
    parameter real      F_min     = 1.4e9    from (0:3e9);
    parameter real      Rth       = 2.1e3    from (0:inf);
    parameter real      tox       = 12e-9    from (0:100e-9);
    parameter real      time_step = 1e-9     from (1e-15: 1);

    real            Vtb, Itb;
    real            T_cur;
    real            gap_ddt;
    real            gamma;
    real            gamma_ini;
    real            gap;

    parameter real      pulse_width = 20n;

    analog begin
        $bound_step(time_step);

        Vtb = V(mem_plus, mem_minus);
        Itb = I(mem_plus, mem_minus);
        gap = V(gap_value);

        T_cur = T_ini + abs( Vtb * Itb * Rth);
        gamma_ini = gamma0;

        if(Vtb < 0) begin
            gamma_ini = 16;
        end

        gamma = gamma_ini - beta * pow((( gap )/1e-9), 3);

        if ((gamma * abs( Vtb )/ tox) < F_min) begin
            gamma = 0;
        end

        gap_ddt = - Vel0 * exp(- q * Ea / kb / T_cur) * sinh(gamma * a0 / tox * q * Vtb / kb / T_cur);

        V(ddt_value) <+ gap_ddt;
    end
endmodule
```

pascalkuthe, 2023-05-04:
this code looks correct at first glance and I am able to reproduce the crash so this is likely a bug in OpenVAF. I will look into the fix. Thanks for reporting and providing an example!

pascalkuthe, 2023-05-09:
I will post a fix for this soon but this model will never do what you want. The problem is that `I(mem_plus, mem_minus)` always returns 0 and is therefore ignored by OpenVAF (hence the crash). While the crash shouldn't happen, it wasn't noticed so far because a current probe that always returns 0 is usually never desired. If you want to make this a voltage probe you can do:

``` systemverilog
V(mem_plus, mem_minus) <+ 0;
```

But this will yield an ideal current probe (so 0 impedance/infinite admittance) so you likely want to add an impedance here aswell.

pascalkuthe, 2023-05-10:
I have fixed the crash. Furthermore, OpemVAF will now create a warning for current probes to branches that always return 0.

PR #67 (2023-05-10): Prepare for release 23.5.0
- Migrate to LLVM 15 and opaque pointers
- fix panic for current probes to trivially open branches (fixes #52, fixes #53, fixes #66)
- warn when using trivially open branches
- add basic logging
- add crash handler
- update build instructions
- bump dependencies
closed 2023-05-10 by pascalkuthe