pascalkuthe/OpenVAF #74: String parameters in OpenVAF

arpadbuermen, 2023-06-30:
Does OpenVaf support string parameters? I tried compiling a .va file containing the following line

(*desc= "Dummy", type="instance", units = "none"*) parameter string dummy = "abc";

OpenVAF crashed (says it panicked because it entered unreachable code). 

On the OSDI side, calling access() for a string parameter returns a void* pointer. How should I interpret this pointer? Should I cast it to char**? 

Suppose I do this for a string parameter: 

char** ptr = (char**)(osdiDescriptor->access(...));
free(*ptr);
*ptr = (char*)malloc(...);
strcpy(*ptr, "new string");

Is this OK in the spirit of OSDI and OpenVAF?

pascalkuthe, 2023-06-30:
thanks for reporting that crash. OpenVAF does support string parameters/variables but because they aren't used much some issues went unnoticed. I fixed the bugs I found in #75 and added a dummy testcase. If you have a real-world model that makes use of string parameters then I would love to add it to the testruit?

Regrding OSDI integration the usage you can indeed cast the pointer to a (`char **`) pointer. Note however that the model will never allocate a string itself. The value is initialized to a null pointer (and only populated with the default value in steup_model/setup_instance  if no value is supplied) so freeing the pointer will likely cause a use after free (unless you ase sure you previously populated the value).

closed 2023-06-30 by pascalkuthe

arpadbuermen, 2023-07-04:
Where is the default value of a string parameter (the one to which the *ptr points) stored? Is it in the .text section as a string constant or is it allocated on the heap at parameter defaulting?

pascalkuthe, 2023-07-04:
It's in text, OSDI models never allocate anything. The only exception is `$display` (and similar functions) because there is no alternative

pascalkuthe, 2023-07-04:
But before you call `setup_model`/`setup_instant` the pointer is just null

pascalkuthe, 2023-07-04:
Adding a `param_given` function sounds reasonable. I dont really want to expose the internal memory lsyout but adding a function makes sense. The intention was that allocations are tracked by the simulator (in an arena allocator for example, usually paramter are just referwnces to the netlist which woild be kept for the entire duration of the simulation). I think a `param_given` functon is useful and has other usecases too so I will add it anyway when I habe tine. I created  #76 to track that.

OoenVAF currently doesn't support vector parameters so that issue hasn't come up yet.

PR #75 (2023-06-30): fix misscompliation of string operations
Fixes #74