pascalkuthe/OpenVAF #76: Expose a `param_given` function in OSDI

pascalkuthe, 2023-07-04:
This allows simulator to determine whether a parameter was manually supplied or set to a default value withku duplicating tracking. This makes freeing string parameters easier for instance. For prior discussion see #74

arpadbuermen, 2023-07-17:
Another suggestion that would be simple to implement. It would be nice to have access to Jacobian entries and RHS residuals without the need to load them into the matrix/RHS vector. These entries are stored in the instance anyway (judging from the sample implementation of a diode model in C from the OSDI docs). It should be sufficient to return the offset of the value of the entry (which is a double) within the instance structure. The following drivers would be nice:

uint32_t resistiveJacobianOffset(uint32_t node1, uint32_t node2);

uint32_t reactiveJacobianOffset(uint32_t node1, uint32_t node2);
(returns UINT32_MAX if reactive entry is not present, i.e. react_ptr_off==UINT32_MAX). 

uint32_t resistiveResidualOffset(uint32_t node);
uint32_t reactiveResidualOffset(uint32_t node);
uint32_t limResistiveResidualOffset(uint32_t node);
uint32_t limReactiveResidualOffset(uint32_t node);

If you don't like driver functions, a table of these offsets in the device descriptor would also be OK. 

One of the possible uses of this extension is the computation of terminal currents. I know Xyce can return terminal currents for all builtin devices.  For that purpose having easy access to the residuals is sufficient. 

Access to Jacobian entries would also open up new uses. For instance, loading of matrix entries by the simulator instead of by the driver function (load_jacobian_*()). I think you need this functionality when using Trilinos as the computational backed.

pascalkuthe, 2023-07-17:
> Another suggestion that would be simple to implement. It would be nice to have access to Jacobian entries and RHS residuals without the need to load them into the matrix/RHS vector. These entries are stored in the instance anyway (judging from the sample implementation of a diode model in C from the OSDI docs). It should be sufficient to return the offset of the value of the entry within the instance structure. The following drivers would be nice:
> 
> uint32_t resistiveJacobianOffset(uint32_t node1, uint32_t node2);
> 
> uint32_t reactiveJacobianOffset(uint32_t node1, uint32_t node2); (returns UINT32_MAX if reactive entry is not present, i.e. react_ptr_off==UINT32_MAX).
> 
> uint32_t resistiveResidualOffset(uint32_t node); uint32_t reactiveResidualOffset(uint32_t node); uint32_t limResistiveResidualOffset(uint32_t node); uint32_t limReactiveResidualOffset(uint32_t node);
> 
> If you don't like driver functions, a table of these offsets in the device descriptor would also be OK.
> 
> One of the possible uses of this extension is the computation of terminal currents. I know Xyce can return terminal currents for all builtin devices. For that purpose having easy access to the residuals is sufficient.
> 
> Access to Jacobian entries would also open up new uses. For instance, loading of matrix entries by the simulator instead of by the driver function (load_jacobian_*()). I think you need this functionality when using Trilinos as the computational backend.

The offsets of the residual are already available in the instance. The matrix pointers are a different matter because:
* there are a lot of then so memory consumption would increase
* I don't really see the usecase here (xyce uses trillios and still stores pointers directly to the matrix directly, it has the option to disable that but it's not used as far as I am aware)