pascalkuthe/OpenVAF #77: Using parameter "--target_cpu" results in error

fmitterb, 2023-07-05:
Using the command line parameter `--target_cpu`, I get the error message shown below.
I have also attached the resulting log file. Version is the nightly build download from July 4th, 2023

If I omit the `--target_cpu` parameter, it works just fine (the resulting osdi library also works as expected in ngspice-40).
The problem also happens for version 23.2.0. The VerilogA file is the Berkeley BSIM-CMG-110 implementation.

I'd like to be able to build 'generic' osdi libraries that I can roll out on our servers while building the osdi library on my more recent local CPU. 

```
$ ./openvaf -V
openvaf 23.5.0

$ ./openvaf --target_cpu generic -o bsimcmg.osdi bsimcmg.va  
OpenVAF encountered a problem and has crashed!

A log file has been generated at "/tmp/openvaf-crash-1688476020.log".

$ cat /tmp/openvaf-crash-1688476020.log

OpenVAF 23.5.0
Panic occurred in file '/root/project/vendor/clap_builder/src/parser/error.rs' at line 32
Mismatch between definition and access of `target_cpu`. Could not downcast to TypeId { t: 13307641874416792075 }, need to downcast to TypeId { t: 9611275717051495212 }
...

$ ./openvaf -o bsimcmg.osdi bsimcmg.va  
Finished building bsimcmg.va in 2.66s
```

[openvaf-crash-1688476020.log](https://github.com/pascalkuthe/OpenVAF/files/11961878/openvaf-crash-1688476020.log)

pascalkuthe, 2023-07-05:
Thanks for reporting. I already fixed the issue in  7587589 after refining your email. I just didn't get around to fixing the download link yet.

If you redownload the newest nightly it should be fixed now.b

fmitterb, 2023-07-06:
Thank you. The new nightly build works for me.

closed 2023-07-06 by fmitterb