pascalkuthe/OpenVAF #78: unexpected token ':'; expected ';'

yrrapt, 2023-07-19:
When attempting to compile the `juncap200rr` model I hit an error:

`error: unexpected token ':'; expected ';'`

I can't post the exact code but the syntax of that line is something like this:

`V(node) : (ddt(V(node)) == (-V(other_node)/CONSTANT + some_other_stuff);`

My knowledge of VerilogA is not good enough to know if that's valid syntax but hopefully it's useful for you.

pascalkuthe, 2023-07-19:
Thanks for reaching out! The snippet you posted is an idirect contribute statement. OpenVAF currently doesn't support indirect contribute statements. Implementing support for those isn't too hard and  its on m todo list for the next release anyway.

In the meantime, you can workaround as follows:

``` verilog
V(node) <+ V(node) +  (ddt(V(node)) - (-V(other_node)/CONSTANT + some_other_stuff);
```

this technically results in incorrect tolerances if you mix different disciplines. However, this is largely a theoretical problem since in practice almost all nodes are electrical. From the language standard:

> However, this approach does not result in the right tolerances being applied to the equation if out and in have
different disciplines. In this situation, the tolerances for the equations would come from V(out) because it is
the target of the contribution, but the final equation does not contain V(out). It would be better if the tolerances for the equation were taken from V(in).
>
> The indirect branch assignment should be used in this situation.

In your example the above workaround should yield the exact same result

yrrapt, 2023-07-22:
Thanks for the speedy response and workaround.

I have used the suggested change the model now compile correctly.

Thanks!

closed 2023-07-22 by yrrapt