pascalkuthe/OpenVAF #83: Parameter Required

kdotom, 2023-08-19:
Are parameters required for a model? I was working on creating a minimal example of something when I came across a potential issue.

This doesn't compile properrly:

```
module testmodule(gnd, o1, i1);
        inout gnd;
        input i1;
        output o1;
        electrical i1, o1, gnd;
 
        //parameter real x = 0;

        analog begin
                V(o1,gnd) <+ 2*V(i1,gnd);
        end
endmodule
```

However, if I create an unused parameter (i.e. uncomment the following line)
```
parameter real x = 0;
```
then the module properly compiles.

I can use this workaround, but just putting it on your radar.

metroid120, 2023-08-21:
I can confirm this issue, it likely is a bug that OpenVAF requires at least 1 model parameter, even if that is not required.

pascalkuthe, 2023-08-21:
I thaught I already fixed this, did you test with the latest nightly release from the README?

Partmedia, 2023-11-07:
I reproduced this bug with Git revision a9697ae77 that I compiled myself from source.

Partmedia, 2023-11-07:
Here is a minimum crashing example:
```
`include "constants.vams"
`include "disciplines.vams"

module foo(a);
    input a;
    electrical a;
    // parameter bar = 0; // uncomment this to not crash
endmodule
```

On my machine this crashes with:
```
Panic occurred in file 'openvaf/mir_llvm/src/context.rs' at line 83
called `Option::unwrap()` on a `None` value
   0..   2: 0x2193584a7306 - core::option::Option<T>::unwrap::ha097196f2facdac9
                at /wrkdirs/usr/ports/lang/rust/work/rustc-1.73.0-src/library/core/src/option.rs:935
                 - mir_llvm::context::CodegenCx::const_str_uninterned::h02627e583b77ff45
                at openvaf/openvaf/mir_llvm/src/context.rs:83
   0: 0x2193583b60e3 - osdi::metadata::osdi_0_3::OsdiNode::to_ll_val::h01997c379c29a1e1
                at openvaf/openvaf/osdi/src/metadata/osdi_0_3.rs:201
   0..   4: 0x21935839b4de - osdi::metadata::osdi_0_3::OsdiDescriptor::to_ll_val::{{closure}}::hb5a656f8b2cf8cbd
                at openvaf/openvaf/osdi/src/metadata/osdi_0_3.rs:320
                 - core::iter::adapters::map::map_fold::{{closure}}::he57f668cd094524f
                at /wrkdirs/usr/ports/lang/rust/work/rustc-1.73.0-src/library/core/src/iter/adapters/map.rs:84
                 - <core::slice::iter::Iter<T> as core::iter::traits::iterator::Iterator>::fold::ha9451af66458d636
                at /wrkdirs/usr/ports/lang/rust/work/rustc-1.73.0-src/library/core/src/slice/iter/macros.rs:232
                 - <core::iter::adapters::map::Map<I,F> as core::iter::traits::iterator::Iterator>::fold::had356e565f2433ab
                at /wrkdirs/usr/ports/lang/rust/work/rustc-1.73.0-src/library/core/src/iter/adapters/map.rs:124

```
Happy to provide full logs if that would be useful.