pascalkuthe/OpenVAF #84: Setting Parameter Values

kdotom, 2023-08-21:
A few models compile, but when I attempt to set the parameters in SPICE, the OSDI model uses the default value as opposed to the value entered in spice.

e.g. for the following module

```
module testmodule(gnd, o1, i1);
        inout gnd;
        input i1;
        output o1;
        electrical i1, o1, gnd;

        // internal nodes
        electrical n1;

        parameter real x = 0;

        analog begin
                V(n1,gnd) <+ x * V(i1,gnd) * (1e-9);
                V(o1,gnd) <+ V(n1,gnd);
        end
endmodule
```
with a spice specification of 
```
.model testmodule_model testmodule(x={5})
```

The simulation result still uses x=0.

Am I missing something or should this work?

metroid120, 2023-08-21:
This is just an ngspice Syntax error. Please see the netlist below with a working example how to use your model:

```
OSDI Example

V1 in  0 DC 0
R2 out 0 1k

* best syntax:
.model testmodule_model testmodule
+ x=1e9

* works too:
*.model testmodule_model testmodule x=1e9

* does not work:
*.model testmodule_model testmodule(x={5})

N1 0 out in testmodule_model

.control
pre_osdi model.osdi

*op
dc V1 0 2 0.01;
plot v(out)

.endc
.end
```

closed 2023-08-21 by metroid120

kdotom, 2023-08-22:
Confirmed. Thanks!