pascalkuthe/OpenVAF #85: requesting license clarification

felix-salfelder, 2023-09-02:
Hi there.

I found a top level LICENSE file with a GPL license in it. This makes believe that the included source code is under the GPLv3+ license. But it appears to be a little more convoluted..

For example, I would expect integration_tests/RESISTOR/resistor.va to be covered by the GPL as there is no indication otherwise. Still, it would be nice to have the license made explicit in the file header. Please add a note there, if you have permission to distribute the file under GPL.

Here's why.

Another file, integration_tests/BSIM4/bsim4.va comes without license note as well, makes a reference to the authors of a related model under a different license (iirc, the C model was released under a BSD license), but does not say that the actual authors (at cogenda) have chosen to impose a CC-BY-NC 4.0 license, which is both BSD and GPL incompatible. This is easy to verify as of today, as the repo still exists [1]. (Note that "NC" means "non-commercial".)

Obviously the author at cogenda failed to include a license statement in their model file, possibly leading to others drawing wrong conclusions. Whoever spent time on their code might end up disappointed and empty-handed. 

I can see that most other files in your repository are lacking a license statement. Assuming that your intent is to contribute (to) free/libre software I would like to ask you to follow FSF recommendation and add a comment *to every single file* that is actually distributable under GPL. Please also include the license version number. A single line is sufficient.

Thanks for your understanding
felix

[1] https://github.com/cogenda/VA-BSIM48/blob/master/README.md

pascalkuthe, 2023-09-02:
Thanks for bringing the missing licenses for bsim3 and bsim to my attention. The only other file with an unclear license is the EKV longchannel model but that is under ECL-2.0 and therefore also GPL3+ compatible (so can be relicensed under GPL3 here). All other integration tests are self-written and therefore fall under GPL.  I have pushed a commit to clarify that and add the missing licenses.

I don't plan to add a disclaimer to the top of every file. That practice is not universal in open source and I consider it noisy (and prone to bitrot). It's also not required from a legal perspective. Everything outside of the integration tests is licensed under GPLv3+.

If you want to be on the save side legally and only use GPLv3+ code simply remove the integration_test directory. Its not required for building openvaf and does not affect the copyright of the final binary in any way (test cases are not used during compilation obviously).

closed 2023-09-02 by pascalkuthe

felix-salfelder, 2023-09-03:
On Sat, Sep 02, 2023 at 08:35:08AM -0700, Pascal Kuthe wrote:
> I have pushed a commit to clarify that and add the missing licenses.

Thanks!