pascalkuthe/OpenVAF #88: analog function ... endfunction of verilog-A is supported in OpenVAF?

xzou0718, 2023-10-09:
Hi Guys,

analog function ... endfunction feature is supported in OpenVAF? 
my verilog-A code list following.
`include "disciplines.vams"                                                                                                                                                          

module my_device (d, g, s, b);
  inout d, g, s, b;
  electrical d, g, s, b;

  real Vgs;

  analog function real calculate_Vgs;
    input Vg;
    input Vs;
    real Vg,Vs;
    begin
      calculate_Vgs = Vg - Vs;
    end
  endfunction

analog begin
V(g,s) <+ calculate_Vgs(V(g), V(s));
end
endmodule

During complied code, no error or warning reported, and then openvaf crashed.
The log info:
A log file has been generated at "/tmp/openvaf-crash-1696825039.log".
To help us fix the problem, please open an issue at https://github.com/pascalkuthe/OpenVAF/
or send an email to pascal.kuthe@semimod.de and attach the log file.
If possible please also attach the source file that OpenVAF was compiling.

dwarning, 2023-10-31:
I can confirm the issue with latest version.
But the problem is not the analog function, openVAF needs still at least one parameter, see #83.

xzou0718, 2024-07-05:
Your comment is very useful, thanks a lot.

closed 2024-07-05 by xzou0718