pascalkuthe/OpenVAF #93: Compiling ucrt.c fails on x86_64 Ubuntu 22.04

MrHighVoltage, 2023-10-31:
Hey,

I get errors while compiling the ucrt.c file that is seemingly required for Windows. It is compiled using the clang-cl driver, which kind of only exists to mimic the MSVC cl compiler interface. With Ubuntu 22.04 on x86_64 I simply can't compile it using this driver, as it always errors out with "no input files". I also tried newer Ubuntu versions, where it also fails to compile on ARM.

```
--- stderr                                                                                                                                  
  $ clang-cl /c /Zl /GS- /clang:--target=x86_64-pc-windows-msvc /clang:-o/OpenVAF/target/release/build/target-56cd26990d4e8a0c/out/ucrt_x64.obj /OpenVAF/openvaf/target/src/ucrt.c                                                                                                          
  clang: error: no input files                                                                                                                                                                                                                                                              
  thread 'main' panicked at 'ucrt compilation succeeds: command exited with non-zero code `clang-cl /c /Zl /GS- /clang:--target=x86_64-pc-windows-msvc /clang:-o/OpenVAF/target/release/build/target-56cd26990d4e8a0c/out/ucrt_x64.obj /OpenVAF/openvaf/target/src/ucrt.c`: 1', openvaf/targ
et/build.rs:42:6                                                                                                                                                                                                                                                                            
  note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```
As it uses the "non-cl-conformant" /clang parameters anyways, I would suggest to change it to directly using clang, which never produced an error for me, which I could get to work with the following command-line:

`clang -c -fno-autolink -fno-stack-protector --target={target}-pc-windows-msvc -o{ucrt_obj} {ucrt_src}`

It should be noted, that I haven't tested the produced windows binary, but the compilation finished without errors on x86_64 and aarch64. The given parameters should be the replacements for the cl-parameters.

pascalkuthe, 2023-10-31:
What clang version are you using? OpenVAF requires a full LLVM toolchain (including clang) to compile. All toolchains fo the llvm toolchain must have the exact same version. Only llvm-15 and llvm-16 are supported (both older and newer version will not work). I would suspect that your clang versions does not match there requirements

clang-cl adds a lot more complex configuration than that. I don't really want to reimplement that here (especially since I am not sure to what degree clang offer stability here, I would rather just take the happy path).

If you have trouble building from source on your distro simply use the docker image (or download the nightly CI builds)

MrHighVoltage, 2023-11-02:
It is somehow strange. I only tested it with LLVM/Clang 15 and 16, but I think I have not yet found the real reason for this compile error. Sometimes it works, sometimes not.

But anyways, it makes sense to keep clang-cl as you said, and I expect it to happen because of a wrong symlink (to a underlying clang call that uses the wrong executeable etc.)  or something like that. I will keep you posted on the progress here.