pascalkuthe/OpenVAF #95: Use of model in proprietary simulator

tcaduser, 2023-11-06:
Hello,

It is unclear what license the generated shared library is under?  I understand there is the license of the original model code, as well as the license of your runtime code?

What is the licensing for the shared library that is being generated, as well as any header files that may need to be included to interface with your API?

metroid120, 2023-11-07:
This is a very good question. After Googling a bit, it seems that there is no general legal consensus. GCC, which is in a similar situation has put the [GCC GPL exception](https://www.gnu.org/licenses/gcc-exception-3.1.en.html) in place. Then it is clear what happens.

I am thus not sure, probably only a lawyer can give a definite answer. It is possible that the compiled binary is considered a derivative work of the model Verilog-A code, and thus must adhere to that license.

felix-salfelder, 2024-04-30:
How about the converse? Say, I implement a (proprietary) model, then compile it with OpenVAF with the intent to distribute the output binary. Assuming that the blob contains bits and pieces from OpenVAF, it's plausible that the OpenVAF copyright holders have their say on the "combined" work.

This section of the GNU [faq](https://www.gnu.org/licenses/gpl-faq.html#WhatCaseIsOutputGPL) may be relevant.

"""The output of a program [here: binary blob] is not, in general, covered by the copyright on the code of the program [here:OpenVAF]. So the license of the code of the program does not apply to the output, whether you pipe it into a file, make a screenshot, screencast, or video."""

What I think matters primarily (also to the original question) is, whether the described use is intended. I recommend to add a clarification somewhere near the license statement.