pascalkuthe/OpenVAF #100: Minimum installation required to run from Windows

tomhajjar, 2023-11-18:
"On Windows the MSVC build tools must be ....installed"

Can someone explain what has to be installed from "VisualStudioSetup.exe" to get the linker function required by OpenVAF? 
I really don't need more unused programs on my PC.

[image]

dwarning, 2023-11-19:
openVAF.exe alone needs the VCRUNTIME140.dll. Look for Visual C++ redistributable package 64-bit. 
If you install the "Desktop development with C++" option from the VS installer you can additional compile and link a very efficient ngspice executable for Windows. Free beer! But is a big bunch.

tomhajjar, 2023-11-20:
I'm still having issues. I have used Qucs 0.0.19 to simulate Verilog-A project examples so I have cursory knowledge of the process.  

I installed "Desktop development with C++" and got the identical errors shown below as before the install. 
I uninstalled "Desktop development with C++" and installed "Visual C++ redistributable package 64-bit" and still have errors.

[image]
[image]

dwarning, 2023-11-20:
You need both: Redistributable package to start openvaf.exe which obviously helped. Please! 
And you need link.exe from Build tools for Visual Studio and perhaps other things like SDK. There are lots of hints in internet doing that e.g. [https://stackoverflow.com/questions/55603111/unable-to-compile-rust-hello-world-on-windows-linker-link-exe-not-found](url). 
It is preferable to use the latest nightly build of openvaf from the Readme of this github.
nigbt_org.va was left over from the former Qucs version <= 0.19. You can't compile it with openVAF.

tomhajjar, 2023-11-20:
Thanks. Got it working.

ra3xdh, 2023-11-21:
See also https://github.com/ra3xdh/qucs_s/issues/373 for solution.