pascalkuthe/OpenVAF #103: Cannot compile if `.va` file does not have `parameter`

mfkasim1, 2023-11-27:
I tried compiling a simple Verilog-A for a capacitor. However, if the file does not have a parameter, it will raise an error. The error can be solved by simply adding a dummy parameter (even though the parameter is useless). Here's an example of the Verilog-A file:
```
`include "disciplines.vams"
`include "constants.vams"

module mycapacitor (p,n);
  electrical p, n;
  inout p, n;

  // parameter real D=1p from (0:inf);  // uncommenting this line would solve the error
  real Q;
  real C = 1p;

  analog
  begin
    Q = C * V(p, n);
    I(p, n) <+ ddt(Q);
  end
endmodule
```
And here's the content of the log file:
```
OpenVAF 23.5.0
Opts {
    defines: [],
    codegen_opts: [],
    lints: [],
    input: "02-mycap.va",
    output: Path {
        lib_file: "02-mycap.osdi",
    },
    include: [],
    opt_lvl: Aggressive,
    target: Target {
        llvm_target: "x86_64-unknown-linux-gnu",
        pointer_width: 64,
        arch: "x86_64",
        data_layout: "e-m:e-p270:32:32-p271:32:32-p272:64:64-i64:64-f80:128-n8:16:32:64-S128",
        options: TargetOptions {
            is_builtin: true,
            cpu: "x86-64",
            features: "",
            linker_flavor: Ld,
            pre_link_args: {
                Ld: [
                    "--no-add-needed",
                    "--hash-style=gnu",
                    "-m",
                    "elf_x86_64",
                ],
            },
            post_link_args: {},
            import_lib: [],
            is_like_windows: false,
            is_like_osx: false,
        },
    },
    target_cpu: "native",
}
Panic occurred in file 'openvaf/mir_llvm/src/context.rs' at line 83
called `Option::unwrap()` on a `None` value
   0: 0x5574cdb423d9 - rust_begin_unwind
                at /rustc/a55dd71d5fb0ec5a6a3a9e8c27b2127ba491ce52/library/std/src/panicking.rs:584
   0: 0x5574cdb65523 - core::panicking::panic_fmt::ha6dc7f2ab2479463
                at /rustc/a55dd71d5fb0ec5a6a3a9e8c27b2127ba491ce52/library/core/src/panicking.rs:142
   0: 0x5574cdb6536d - core::panicking::panic::hb3ad04c589a0e3c8
                at /rustc/a55dd71d5fb0ec5a6a3a9e8c27b2127ba491ce52/library/core/src/panicking.rs:48
   0..   2: 0x5574cba107c6 - core::option::Option<T>::unwrap::h47c9143d060d0350
                at /rustc/a55dd71d5fb0ec5a6a3a9e8c27b2127ba491ce52/library/core/src/option.rs:775
                 - mir_llvm::context::CodegenCx::const_str_uninterned::hcabc88e94ae421e5
[... 75 lines cut ...]
   0: 0x5574cb79cd84 - std::panic::catch_unwind::ha2733c84503a87ca
                at /rustc/a55dd71d5fb0ec5a6a3a9e8c27b2127ba491ce52/library/std/src/panic.rs:137
   0: 0x5574cb7bf4c4 - rayon_core::unwind::halt_unwinding::h518e83fcc0f0f091
                at /root/project/vendor/rayon-core/src/unwind.rs:17
   0..   2: 0x5574cb79e4ab - rayon_core::job::JobResult<T>::call::h526ca128ef426e44
                at /root/project/vendor/rayon-core/src/job.rs:218
                 - <rayon_core::job::StackJob<L,F,R> as rayon_core::job::Job>::execute::hc5e53551636bbb56
                at /root/project/vendor/rayon-core/src/job.rs:120
   0..   3: 0x5574cb7f8df5 - rayon_core::job::JobRef::execute::hd11389873f126f2b
                at /root/project/vendor/rayon-core/src/job.rs:64
                 - rayon_core::registry::WorkerThread::execute::hd471ca85cef13df3
                at /root/project/vendor/rayon-core/src/registry.rs:874
                 - rayon_core::registry::WorkerThread::wait_until_cold::h7b7483b3044a671d
                at /root/project/vendor/rayon-core/src/registry.rs:820
   0..   3: 0x5574cb7f68e9 - rayon_core::registry::WorkerThread::wait_until::h738b3a6e29f6f187
                at /root/project/vendor/rayon-core/src/registry.rs:803
                 - rayon_core::registry::main_loop::h326959b63dfe9b95
                at /root/project/vendor/rayon-core/src/registry.rs:948
                 - rayon_core::registry::ThreadBuilder::run::he1cd9f5104020c8e
                at /root/project/vendor/rayon-core/src/registry.rs:54
   0: 0x5574cb80037a - <rayon_core::registry::DefaultSpawn as rayon_core::registry::ThreadSpawn>::spawn::{{closure}}::hcc83daca173cc139
                at /root/project/vendor/rayon-core/src/registry.rs:99
                 - std::sys_common::backtrace::__rust_begin_short_backtrace::h1c91c9e680a6d48d
                at /rustc/a55dd71d5fb0ec5a6a3a9e8c27b2127ba491ce52/library/std/src/sys_common/backtrace.rs:122
```

gjcoram, 2024-05-23:
This is kind of weird. From the stack trace, the error seems to be coming from OsdiNode::to_ll_val

5: 0x5574cb7df3c6 - osdi::metadata::osdi_0_3::OsdiNode::to_ll_val::h8dc42151ca46ac19
                at /root/project/openvaf/osdi/src/metadata/osdi_0_3.rs:216

Line 216 is converting the residual_units of the OsdiNode:
    ctx.const_str_uninternetd(&self.residual_units),

Why is this a problem when there is no parameter in the module?

gjcoram, 2024-05-23:
% git diff openvaf/mir_llvm/src/context.rs
diff --git a/openvaf/mir_llvm/src/context.rs b/openvaf/mir_llvm/src/context.rs
index cf4f71f..953d825 100644
--- a/openvaf/mir_llvm/src/context.rs
+++ b/openvaf/mir_llvm/src/context.rs
@@ -80,7 +80,7 @@ impl<'a, 'll> CodegenCx<'a, 'll> {
     }
 
     pub fn const_str_uninterned(&self, lit: &str) -> &'ll Value {
-        let lit = self.literals.get(lit).unwrap();
+        let lit = self.literals.get(lit).unwrap_or_default();
         self.const_str(lit)
     }