pascalkuthe/OpenVAF #105: `laplace_*` feature request (willing to contribute, just need pointer)

mfkasim1, 2023-11-28:
Hi, I would like to use the `laplace_*` (e.g., `laplace_nd`, etc) for my work. As it's not implemented in OpenVAF, I'm willing to put my time and effort if it helps you in implementing the features (I have several contribution experience to large open source software, like numpy). Could you please give me some pointer on where to look first to implement the feature?

felix-salfelder, 2024-03-01:
We now have an initial laplace_nd filter in the recent modelgen snapshot [1]. The implementation might give some ideas, in case anyone would like to add it to OpenVAF.

[1] https://git.savannah.gnu.org/cgit/gnucap/gnucap-modelgen-verilog.git/tag/?h=20240229-dev

kdotom, 2024-05-28:
How is the laplace_nd development going? Any help needed?