pascalkuthe/OpenVAF #107: Incorrect handling of $mfactor

arpadbuermen, 2023-12-15:
An inductor defined as 

```
`include "constants.vams"
`include "disciplines.vams"
module inductor(A,B);
	inout A, B;
	electrical A, B;
	branch (A,B) br;
	(*desc= "Inductance", type="instance", units = "H"*) parameter real l = 1e-6 from [0:inf];
	(*desc= "Flux", units ="Vs" *) real flux;
	real Phi;
	analog begin
                Phi = l * I(br);
                V(br) <+ ddt(Phi);
                flux = Phi;
	end
endmodule
```
Does not take into account $mfactor correctly. If I create a circuit that drives 1A through a 2mH inductor with $mfactor=2 it results in I(br)=0.25 instead of 0.5. For $mfactor=3 I get I(br)=0.1111. 

If I create a resistor in Verilog-A the $mfactor is handled correctly. 

Seems that for flow nodes the $mfactor is somehow applied twice.

gjcoram, 2024-07-08:
How are you measuring I(br)? Are you using $strobe to report the value, or declaring an op-pt value?

arpadbuermen, 2024-07-09:
Flow branches were incorrectly handled when $mfactor was not 1. I think it is somewhere in the LRM that current in an instance with $mfactor!=1 should reflect the value of a single parallel instance (The value returned by any branch flow probe in the analog block, including those used in indirect assignments, shall be divided by $mfactor). 

Branch flow probes in OpenVAF return the value of the corresponding flow unknown. So a flow unknown should reflect the current of a single parallel instance. This makes sense because whoever does the probing does not have to know the $mfactor of the instance within which the probed current resides. This also makes sense because you can write a compact model without thinking about $mfactor and then if you create an instance with $mfactor!=1 everything works out of the box. 

I stumbled upon this bug when I was testing the Verilog-A model of an inductor in VACASK. The system of equations is now such that the unknown corresponding to the flow (current) of an inductor refers to the current of a single parallel inductor instance. Same goes for a voltage source (where a flow unknown is added for each voltage source).