pascalkuthe/OpenVAF #108: unable to use verilogae

tcaduser, 2023-12-19:
I am trying to use verilogae, using the syntax presented here:
https://man.sr.ht/~dspom/openvaf_doc/verilogae/examples.md

However this doesn't seem to work from a build using your installation instructions here:
https://github.com/pascalkuthe/OpenVAF?tab=readme-ov-file#building-openvaf-with-docker

For example:
```
~/git/OpenVAF/target/release/verilogae  install ekv26_SDext_Verilog-A.va 
thread 'main' panicked at 'called `Result::unwrap()` on an `Err` value: failed to read install

Caused by:
    No such file or directory (os error 2)', verilogae/verilogae/src/main.rs:6:48
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```

```
RUST_BACKTRACE=1 ~/git/OpenVAF/target/release/verilogae  install ekv26_SDext_Verilog-A.va 
thread 'main' panicked at 'called `Result::unwrap()` on an `Err` value: failed to read install

Caused by:
    No such file or directory (os error 2)', verilogae/verilogae/src/main.rs:6:48
stack backtrace:
   0: rust_begin_unwind
             at /rustc/897e37553bba8b42751c67658967889d11ecd120/library/std/src/panicking.rs:584:5
   1: core::panicking::panic_fmt
             at /rustc/897e37553bba8b42751c67658967889d11ecd120/library/core/src/panicking.rs:142:14
   2: core::result::unwrap_failed
             at /rustc/897e37553bba8b42751c67658967889d11ecd120/library/core/src/result.rs:1785:5
   3: core::result::Result<T,E>::unwrap
             at /rustc/897e37553bba8b42751c67658967889d11ecd120/library/core/src/result.rs:1107:23
   4: verilogae::main
             at /io/verilogae/verilogae/src/main.rs:6:5
   5: core::ops::function::FnOnce::call_once
             at /rustc/897e37553bba8b42751c67658967889d11ecd120/library/core/src/ops/function.rs:248:5
note: Some details are omitted, run with `RUST_BACKTRACE=full` for a verbose backtrace.
```

What is the best way to get verilogae to work?  I do not see there to be a download available.

metroid120, 2023-12-19:
Hi, unfortunately, the documentation is very outdated. Nobody is actively working on the project right now - unfortunately.
Still, the code works well, let me give you some guidance. It only works on Linux, though.

First, you must install verilogae in your Python. You have two options.
1) Try to run:
`pip install verilogae`
It will download and install from here https://pypi.org/project/verilogae/#files

2) If this does not work, you can compile yourself. Use the Dockerfile I attached below. 
Run
`docker build -t vae_build . `
Then 
`docker run -ti -v $(pwd):/in vae_build:latest  `
to mount your current directory of the local machine in the Docker and you can find the compiled wheel files 
under /io/OpenVAF/wheels.
These wheels can be directly installed in Python.

Now, this is how you load the model in Python:
```
import verilogae
hl2 = verilogae.load(path_to_model.va)
```
Hope it helps!

tcaduser, 2023-12-19:
Thanks.  I am now able to start experimenting with the pypi package.