pascalkuthe/OpenVAF #123: error: no member named 'setHasAllowReassoc'

yurivict, 2024-04-26:
Build can't find a symbol in llvm:
```
 cargo:warning=wrapper/OpenVafWrapper.cpp:23:8: error: no member named 'setHasAllowReassoc' in 'llvm::Instruction'; did you mean 'setHasAllowReciprocal'?
  cargo:warning=   23 |     I->setHasAllowReassoc(true);
  cargo:warning=      |        ^~~~~~~~~~~~~~~~~~
  cargo:warning=      |        setHasAllowReciprocal
```

Version: 23.5.0
llvm-16 and llvm-17
FreeBSD 14.0

metroid120, 2024-04-26:
OpenVAF can only be built with llvm-15, this is known and stated in the Readme. For building, use the Docker container.

yurivict, 2024-04-26:
When I am trying to build with llvm-15 it fails in a different way:
```
[target 0.0.0] cargo:rustc-env=CFG_COMPILER_HOST_TRIPLE=x86_64-unknown-freebsd
[target 0.0.0] cargo:rerun-if-env-changed=RUST_CHECK
[target 0.0.0] cargo:rerun-if-changed=/usr/ports/cad/openvaf/work/OpenVAF-OpenVAF-v23.5.0/openvaf/target/src/ucrt.c
[target 0.0.0] clang-5.0: error: no such file or directory: '/clang:--target=x86_64-pc-windows-msvc'
[target 0.0.0] clang-5.0: error: no such file or directory: '/clang:-o/usr/ports/cad/openvaf/work/target/x86_64-unknown-freebsd/release/build/target-58917fcd0acd8a5e/out/ucrt_x64.obj'
[target 0.0.0] thread 'main' panicked at openvaf/target/build.rs:42:6:
[target 0.0.0] ucrt compilation succedes: command exited with non-zero code `clang-cl /c /Zl /GS- /clang:--target=x86_64-pc-windows-msvc /clang:-o/usr/ports/cad/openvaf/work/target/x86_64-unknown-freebsd/release/build/target-58917fcd0acd8a5e/out/ucrt_x64.obj /usr/ports/cad/openvaf/work/OpenVAF-OpenVAF-v23.5.0/openvaf/target/src/ucrt.c`: 1
[target 0.0.0] stack backtrace:
[target 0.0.0]    0: rust_begin_unwind
[target 0.0.0]    1: core::panicking::panic_fmt
[target 0.0.0]    2: core::result::unwrap_failed
[target 0.0.0]    3: core::result::Result<T,E>::expect
[target 0.0.0]              at /wrkdirs/usr/ports/lang/rust/work/rustc-1.77.0-src/library/core/src/result.rs:1030:23
[target 0.0.0]    4: build_script_build::gen_msvcrt_importlib
[target 0.0.0]              at ./build.rs:37:5
[target 0.0.0]    5: build_script_build::main
[target 0.0.0]              at ./build.rs:14:5
[target 0.0.0]    6: core::ops::function::FnOnce::call_once
[target 0.0.0]              at /wrkdirs/usr/ports/lang/rust/work/rustc-1.77.0-src/library/core/src/ops/function.rs:250:5
[target 0.0.0] note: Some details are omitted, run with `RUST_BACKTRACE=full` for a verbose backtrace.
error: failed to run custom build command for `target v0.0.0 (/usr/ports/cad/openvaf/work/OpenVAF-OpenVAF-v23.5.0/openvaf/target)`

Caused by:
  process didn't exit successfully: `CARGO=/usr/local/bin/cargo CARGO_CFG_PANIC=unwind CARGO_CFG_TARGET_ARCH=x86_64 CARGO_CFG_TARGET_ENDIAN=little CARGO_CFG_TARGET_ENV='' CARGO_CFG_TARGET_FAMILY=unix CARGO_CFG_TARGET_FEATURE=fxsr,sse,sse2 CARGO_CFG_TARGET_HAS_ATOMIC=16,32,64,8,ptr CARGO_CFG_TARGET_OS=freebsd CARGO_CFG_TARGET_POINTER_WIDTH=64 CARGO_CFG_TARGET_VENDOR=unknown CARGO_CFG_UNIX='' CARGO_ENCODED_RUSTFLAGS='-Clink-arg=-fstack-protector-strong' CARGO_MANIFEST_DIR=/usr/ports/cad/openvaf/work/OpenVAF-OpenVAF-v23.5.0/openvaf/target CARGO_PKG_AUTHORS=DSPOM CARGO_PKG_DESCRIPTION='' CARGO_PKG_HOMEPAGE='' CARGO_PKG_LICENSE=GPL-3.0 CARGO_PKG_LICENSE_FILE='' CARGO_PKG_NAME=target CARGO_PKG_README=../README.md CARGO_PKG_REPOSITORY='' CARGO_PKG_RUST_VERSION=1.56 CARGO_PKG_VERSION=0.0.0 CARGO_PKG_VERSION_MAJOR=0 CARGO_PKG_VERSION_MINOR=0 CARGO_PKG_VERSION_PATCH=0 CARGO_PKG_VERSION_PRE='' DEBUG=true HOST=x86_64-unknown-freebsd LD_LIBRARY_PATH='/usr/ports/cad/openvaf/work/target/release/deps:/usr/ports/cad/openvaf/work/target/release:/usr/local/lib/rustlib/x86_64-unknown-freebsd/lib' NUM_JOBS=7 OPT_LEVEL=3 OUT_DIR=/usr/ports/cad/openvaf/work/target/x86_64-unknown-freebsd/release/build/target-58917fcd0acd8a5e/out PROFILE=release RUSTC=/usr/local/bin/rustc RUSTC_LINKER=/usr/local/llvm15/bin/clang RUSTDOC=/usr/local/bin/rustdoc TARGET=x86_64-unknown-freebsd /usr/ports/cad/openvaf/work/target/release/build/target-e2695bfebac664ce/build-script-build` (exit status: 101)
  --- stdout
  cargo:rustc-env=CFG_COMPILER_HOST_TRIPLE=x86_64-unknown-freebsd
  cargo:rerun-if-env-changed=RUST_CHECK
  cargo:rerun-if-changed=/usr/ports/cad/openvaf/work/OpenVAF-OpenVAF-v23.5.0/openvaf/target/src/ucrt.c

  --- stderr
  $ clang-cl /c /Zl /GS- /clang:--target=x86_64-pc-windows-msvc /clang:-o/usr/ports/cad/openvaf/work/target/x86_64-unknown-freebsd/release/build/target-58917fcd0acd8a5e/out/ucrt_x64.obj /usr/ports/cad/openvaf/work/OpenVAF-OpenVAF-v23.5.0/openvaf/target/src/ucrt.c
  clang-5.0: error: no such file or directory: '/clang:--target=x86_64-pc-windows-msvc'
  clang-5.0: error: no such file or directory: '/clang:-o/usr/ports/cad/openvaf/work/target/x86_64-unknown-freebsd/release/build/target-58917fcd0acd8a5e/out/ucrt_x64.obj'
  thread 'main' panicked at openvaf/target/build.rs:42:6:
  ucrt compilation succedes: command exited with non-zero code `clang-cl /c /Zl /GS- /clang:--target=x86_64-pc-windows-msvc /clang:-o/usr/ports/cad/openvaf/work/target/x86_64-unknown-freebsd/release/build/target-58917fcd0acd8a5e/out/ucrt_x64.obj /usr/ports/cad/openvaf/work/OpenVAF-OpenVAF-v23.5.0/openvaf/target/src/ucrt.c`: 1
  stack backtrace:
     0: rust_begin_unwind
     1: core::panicking::panic_fmt
     2: core::result::unwrap_failed
     3: core::result::Result<T,E>::expect
               at /wrkdirs/usr/ports/lang/rust/work/rustc-1.77.0-src/library/core/src/result.rs:1030:23
     4: build_script_build::gen_msvcrt_importlib
               at ./build.rs:37:5
     5: build_script_build::main
               at ./build.rs:14:5
     6: core::ops::function::FnOnce::call_once
               at /wrkdirs/usr/ports/lang/rust/work/rustc-1.77.0-src/library/core/src/ops/function.rs:250:5
  note: Some details are omitted, run with `RUST_BACKTRACE=full` for a verbose backtrace.
warning: build failed, waiting for other jobs to finish...
warning: `rowan` (lib) generated 1 warning
warning: `syn` (lib) generated 1 warning
*** Error code 101
```

It uses ```windows-msvc``` arguments on FreeBSD.

--------------

Docker doesn't work on FreeBSD.