pascalkuthe/OpenVAF #125: issue with docker setup

gjcoram, 2024-05-22:
When I try this command as part of the instructions for building OpenVAF:

docker run -ti -v .\:/io ghcr.io/pascalkuthe/ferris_ci_build_x86_64-unknown-linux-gnu:latest

I get an error:

bash: /usr/libexec/grepconf.sh: No such file or directory

I think it would be sufficient to create this as an empty file.

dwarning, 2024-05-22:
I have to use "sudo" before both docker cmd's.
Got same error of missing grepconf.sh
But: Then I can follow the next cmd's with successful build of the binaries.

grepconf.sh makes only colors for warnings and errors in compiling process. Seems redhat/centos.

metroid120, 2024-05-29:
I would recommend to not use the Dockerfile linked here, but check out the one used by Harald Pretl
https://github.com/iic-jku/osic-multitool/tree/main/openvaf