pascalkuthe/OpenVAF #127: location of discipline.h and constants.h

holmesozic, 2024-05-29:
`include "discipline.h"
`include "constants.h"

Where are these files located? Are the compiled into the openvaf executable? Are the in the git repo? I would like to copy-modify them and point the compiler to the modified version using:

openvaf -I ~/myinclude

holmesozic, 2024-05-29:
I have part of the answer with:
openvaf --print-expansion -o thing.osdi thing.va 2>thing.error

I guess I can get the originals here?

// Copyright(c) 2009-2014 Accellera Systems Initiative Inc.
// 1370 Trancas Street #163, Napa, CA 94558, USA.
//
// The material in disciplines.vams is an essential part of the Accellera Systems
// Initiative ("Accellera") Verilog-AMS Language Standard. Verbatim copies of
// the material in this Annex may be used and distributed without restriction.
// All other uses require permission from Accellera IP Committee
// (ipr-chair@lists.accellera.org).
// All other rights reserved.
//
// Version 2.4.0

gjcoram, 2024-05-31:
Yes, you should get the originals from Accellera. Most open-source licenses allow modification, but Accellera only allows "verbatim copies" of these standard header files to be distributed, so these files can't be included in open-source packages.