pascalkuthe/OpenVAF #135: singular matrix with voltage contribution

gjcoram, 2024-07-05:
When I have a voltage branch contribution that connects implicitly to ground:
    V(a) <+ dc;
OpenVAF does not properly construct the circuit matrix, such that I get a singular matrix error from ngspice.

If I change this to
    V(a,b) <+ dc;
and connect 'b' to ground in the netlist, it works fine.

gjcoram, 2024-07-05:
[v_contrib.zip](https://github.com/user-attachments/files/16113139/v_contrib.zip)
v_contrib.zip contains:
v_contrib_1.va v_contrib_1.sp -- shows singular matrix
v_contrib_2.va v_contrib_2.sp -- analysis runs properly

gjcoram, 2024-07-05:
I'm still new at Rust, but I am curious about the function below, where 3 of the 5 cases have "cursor.ins()". Case 1, we're adding 0 to something (I think "dst" is "destination"), and we don't need anything there. But in case 3, (F_ZERO, _) when negate is false, should there be a cursor.ins()? Without it, the variable that's used for KCL in the matrix stamp is dropped (it does not seem to be marked as being used, and thus is not in the "live_params" in sim_unknown_reads that are considered when building the jacobian.

`pub fn add(cursor: &mut FuncCursor, dst: &mut Value, val: Value, negate: bool) {
    match (*dst, val) {
        (_, F_ZERO) => (),
        (F_ZERO, _) if negate => *dst = cursor.ins().fneg(val),
        (F_ZERO, _) => *dst = val,
        (old, _) if negate => *dst = cursor.ins().fsub(old, val),
        (old, _) => *dst = cursor.ins().fadd(old, val),
    }
}
`

georgtree, 2024-07-06:
Hello, I also faced with this issue, and the only solution is to add explicit ground reference everywhere.

gjcoram, 2024-07-09:
> Hello, I also faced with this issue, and the only solution is to add explicit ground reference everywhere.

The commit I made solves the issue.

georgtree, 2024-07-09:
I will check it out, thank you.
I have found another issue, would it better to add it here or at your fork?
Thank you.