pascalkuthe/OpenVAF #145: ARM64 Installation

shaurya0406, 2024-09-29:
I wanted to install OpenVAF on my `Ubuntu 22.04.02 ARM64 VM` running on a Mac (M2 Pro) machine via Parallels.
Is there any guide available for that?
Thank you!

adrienluitot, 2024-11-24:
Hello, I started an install on Ubuntu 24.04 arm64, I achieved to compile it.
(However I use LLVM 18, and I had to use @Kreijstal llvm version of OpenVAF.)
It kinda works, so you should be able to do it too.

First you need the dependencies: 
`sudo apt install clang clang-tools llvm lld cargo`

Then you can clone the repo, and just build OpenVAF:
`cargo build --release`

If you have errors, send them here, so we can help you.

shaurya0406, 2024-11-28:
Thanks a lot!

closed 2024-11-28 by shaurya0406