pascalkuthe/OpenVAF #146: Crashed while trying to compile Verilog-A Module

thomaspepler, 2024-10-02:
I was trying to compile the attached basic Verilog-A module (vco.va) and OpenVAF crashed. The crash log is also attached. I added a .txt extension to be able to attach the files.

For what it's worth, the line in the log file "index out of bounds: the len is 2 but the index is 67108863" suggests the index was set to 0x3FF_FFFF.
[vco.va.txt](https://github.com/user-attachments/files/17219207/vco.va.txt)
[openvaf-crash-1727835402.log.txt](https://github.com/user-attachments/files/17219208/openvaf-crash-1727835402.log.txt)

dwarning, 2024-10-02:
A crash should never happen, but the reason is the unsupported $bound_step() function. openvaf was developed mainly with the focus on compact models and is not covering the entire language syntax.

thomaspepler, 2024-10-16:
Thank you for the response. Is there any possibility of implementing $bound_step(), or would it require more integration with the simulator's time step engine, e.g. ngspice? Could I contribute in any way?

felix-salfelder, 2024-10-17:
Hi Thomas.

Thanks for the example.

> Could I contribute in any way?

The slightly modified vco works with Gnucap [1]. We are interested in an OSDI wrapper, it could serve as a starting point for further OpenVAF experiments. Depends on what you are trying to achieve.

(NB: I'm not familiar with OpenVAF, but I know a little about SPICE and Verilog-A).

[1] https://git.savannah.gnu.org/cgit/gnucap/gnucap-modelgen-verilog.git/commit/?h=develop&id=dc515985a94b66c843defa22e6266f7cf82f4001

Lang0722, 2024-10-18:
> Hi Thomas.
>
> Thanks for the example.
>
>
> The slightly modified vco works with Gnucap [1]. We are interested in an OSDI wrapper, it could serve as a starting point for further OpenVAF experiments. [...]

Have your investigate the implementation of C code modle or MIF devices in NGspice's Xspice? I think they can aslo dynamically load the model. But to compatiable with VA, more features need to be implemented.

felix-salfelder, 2024-10-18:
> Have your investigate the implementation of C code modle or MIF devices in NGspice's Xspice?

Dear Lang.

Not sure what you are asking for. It may well be possible to represent the vco as an Xspice extension for NGspice. There is still so much lower hanging fruit...