pascalkuthe/OpenVAF #150: OpenVAF has crashed while compiling model with optical discipline

KozhevnikovDA, 2025-01-16:
Trying to compile Verilog-A model with optical discipline and got an error “OpenVAF encountered a problem and has crashed!” with log. Problem seems to be in “E(outP2C,gnd) <+ V(ampin);” part.

```
`include "disciplines.vams"
 
// Optical
nature Efield
    units  = "E";
    access = E;
endnature

// Conservative discipline
discipline optical
    potential Efield;
enddiscipline

module laser_cw(gnd, out, Htemp, ampin);

    inout gnd;
    input Htemp; 
    input ampin; 
    output out;

    electrical ampin;
    thermal Htemp;
    optical gnd;
    optical out;

    // Internal nodes
    optical outP2C;

    // Device parameters
    parameter real gain=1e10;  // gain of temperature to frequency
    parameter real offsetfreq=0;  // offset from center frequency

    analog begin
        // magnitude
        E(outP2C,gnd) <+ V(ampin);
        // phase
        // ...
    end

endmodule
```

log:
```
OpenVAF 23.5.0
Opts {
    defines: [],
    codegen_opts: [],
    lints: [],
    input: "laser_cw.va",
    output: Path {
        lib_file: "laser_cw.osdi",
    },
    include: [],
    opt_lvl: Aggressive,
    target: Target {
        llvm_target: "x86_64-pc-windows-msvc",
        pointer_width: 64,
        arch: "x86_64",
        data_layout: "e-m:w-p270:32:32-p271:32:32-p272:64:64-i64:64-f80:128-n8:16:32:64-S128",
        options: TargetOptions {
            is_builtin: true,
            cpu: "x86-64",
            features: "",
            linker_flavor: Msvc,
            pre_link_args: {
                Msvc: [
                    "/NOLOGO",
                ],
            },
            post_link_args: {
                Msvc: [
                    "msvcrt.lib",
                ],
            },
            import_lib: [
                33,
                60,
                97,
				
            _______ a lot of numbers____
			
                102,
                0,
                10,
            ],
            is_like_windows: true,
            is_like_osx: false,
        },
    },
    target_cpu: "native",
}
Panic occurred in file 'openvaf/osdi/src/metadata.rs' at line 352
called `Option::unwrap()` on a `None` value
   0: 0x7ff7ecdde613 - <unresolved>
   0: 0x7ff7ecd58989 - <unresolved>
   0: 0x7ff7ecd540d4 - <unresolved>
   0: 0x7ff7ecdb7e22 - <unresolved>
   0: 0x7ff7ecdb7b82 - <unresolved>
   0: 0x7ff7ecdb56f7 - <unresolved>
   0: 0x7ff7ecdb7899 - <unresolved>
   0: 0x7ff7eea9a575 - <unresolved>
   0: 0x7ff7eea9a41c - <unresolved>
   0: 0x7ff7ece14bd8 - <unresolved>
   0: 0x7ff7ecf4ab05 - <unresolved>
   0: 0x7ff7ece0ee0b - <unresolved>
   0: 0x7ff7ecf32a68 - <unresolved>
   0: 0x7ff7ece15118 - <unresolved>
   0: 0x7ff7ecd5b086 - <unresolved>
   0: 0x7ff7ecd51743 - <unresolved>
   0: 0x7ff7ecd4f826 - <unresolved>
   0: 0x7ff7ecd5665c - <unresolved>
   0: 0x7ff7ecda722b - <unresolved>
   0: 0x7ff7ecd56640 - <unresolved>
   0: 0x7ff7eea987c8 - <unresolved>
   0: 0x7ff9b008259d - BaseThreadInitThunk
   0: 0x7ff9b0eeaf38 - RtlUserThreadStart
```

[laser_cw.txt](https://github.com/user-attachments/files/18435502/laser_cw.txt)

[openvaf-crash-1737013732.log](https://github.com/user-attachments/files/18435469/openvaf-crash-1737013732.log)

Kreijstal, 2025-01-16:
problem seems to be 

```
called `Option::unwrap()` on a `None` value
```

```rust

    // its valid to have disciplines without pot/flow nature but then we can't
    // have branches for those so its ok to unwrap here
    let units = discipline
        .map(|discipline| {
            let nature = if is_flow {
                discipline.flow(db).unwrap()   //<---- unwrap here
            } else {
                discipline.potential(db).unwrap()
            };
            nature.units(db)
        })
        .unwrap_or_default();

    (name, units, is_flow)

```

Kreijstal, 2025-01-16:
Feel free to report to https://github.com/arpadbuermen/OpenVAF

KozhevnikovDA, 2025-01-22:
But there is a potential nature in discipline, isn't it?

```
// Conservative discipline
discipline optical
    potential Efield;
enddiscipline
```

KozhevnikovDA, 2025-01-22:
The same error result even if nature is flow

Kreijstal, 2025-01-22:
> The same error result even if nature is flow

post the error

KozhevnikovDA, 2025-01-22:
```
OpenVAF 23.5.0
Opts {
    defines: [],
    codegen_opts: [],
    lints: [],
    input: "laser2.va",
    output: Path {
        lib_file: "laser2.osdi",
    },
    include: [],
    opt_lvl: Aggressive,
    target: Target {
        llvm_target: "x86_64-pc-windows-msvc",
        pointer_width: 64,
        arch: "x86_64",
        data_layout: "e-m:w-p270:32:32-p271:32:32-p272:64:64-i64:64-f80:128-n8:16:32:64-S128",
        options: TargetOptions {
            is_builtin: true,
            cpu: "x86-64",
            features: "",
            linker_flavor: Msvc,
            pre_link_args: {
                Msvc: [
                    "/NOLOGO",
                ],
            },
            post_link_args: {
                Msvc: [
                    "msvcrt.lib",
                ],
            },
            import_lib: [
                33,
                60,
                97,
             ...
___________________________________________
             ...
                0,
                10,
            ],
            is_like_windows: true,
            is_like_osx: false,
        },
    },
    target_cpu: "native",
}
Panic occurred in file 'openvaf/osdi/src/metadata.rs' at line 352
called `Option::unwrap()` on a `None` value
   0: 0x7ff702e8e613 - <unresolved>
   0: 0x7ff702e08989 - <unresolved>
   0: 0x7ff702e040d4 - <unresolved>
   0: 0x7ff702e67e22 - <unresolved>
   0: 0x7ff702e67b82 - <unresolved>
   0: 0x7ff702e656f7 - <unresolved>
   0: 0x7ff702e67899 - <unresolved>
   0: 0x7ff704b4a575 - <unresolved>
   0: 0x7ff704b4a41c - <unresolved>
   0: 0x7ff702ec4bd8 - <unresolved>
   0: 0x7ff702ffab05 - <unresolved>
   0: 0x7ff702ebee0b - <unresolved>
   0: 0x7ff702fe2a68 - <unresolved>
   0: 0x7ff702ec5118 - <unresolved>
   0: 0x7ff702e0b086 - <unresolved>
   0: 0x7ff702e01743 - <unresolved>
   0: 0x7ff702dff826 - <unresolved>
   0: 0x7ff702e0665c - <unresolved>
   0: 0x7ff702e5722b - <unresolved>
   0: 0x7ff702e06640 - <unresolved>
   0: 0x7ff704b487c8 - <unresolved>
   0: 0x7ff9b008259d - BaseThreadInitThunk
   0: 0x7ff9b0eeaf38 - RtlUserThreadStart
```

[openvaf-crash-1737541503.log](https://github.com/user-attachments/files/18504046/openvaf-crash-1737541503.log)

KozhevnikovDA, 2025-02-07:
For those who are interested: сompilation is possible if discipline contains both natures - potential and flow

closed 2025-02-07 by KozhevnikovDA