pascalkuthe/OpenVAF #152: issue with 'type' parameter being ignored in transistor verilog-a model

nzagni91, 2025-02-19:
When trying to simulate a CMOS inverter in ngspice with the MVS verilog-A model available on nanohub I run into a peculiar issue with a parameter 'type' in the code that discriminates between the NMOS and PMOS.
Basically, the 'type' parameter is ignored! So when I included the model card the PMOS was behaving exactly as the NMOS, as it is the default.

I solved by renaming the 'type' parameter to 'tipe'. 

Please be wary that also BSIM uses 'type' to define NMOS or PMOS, so it will also be affected!

Kreijstal, 2025-02-19:
> When trying to simulate a CMOS inverter in ngspice with the MVS verilog-A model available on nanohub I ran into a peculiar issue with a parameter 'type' in the code that discriminates between the NMOS and PMOS. Basically, the 'type' parameter is ignored! So when I included the model card the PMOS was behaving exactly as the NMOS, as it is the default.
> 
> I solved by renaming the 'type' parameter to 'tipe'.
> 
> Please be wary that also BSIM uses 'type' to define NMOS or PMOS, so it will also be affected!

please provide a MWE and report it on arpad's fork

dwarning, 2025-02-19:
Can you share your testcase including modelcards.

I have no problem with the provided testbench, even it looks a bit strange:

[image]

nzagni91, 2025-02-19:
Yes I can share the testcase, but to be fair I was having the same issue with the example #2 on openvaf (https://openvaf.semimod.de/docs/getting-started/examples/). Basically with the provided code the inverter was not working due to the same issue.

I am using Windows 11.

nzagni91, 2025-02-19:
The zip file contains a working example of the MVS-based inverter. I had to change the verilog code 'type' parameter to have it run.
[example_mvs.zip](https://github.com/user-attachments/files/18870321/example_mvs.zip)

dwarning, 2025-02-19:
You are using version 1.1, the old one.
I made your changes back: tipe -> type and got reasonable inverter results.
My openvaf version is 23.5.0.