pascalkuthe/OpenVAF #155: which repository is the reference ?

jalcim, 2025-03-08:
I would like to know which OpenVAF repository is the reference, since the original one has not been maintained for two years.

this one ?
https://github.com/chengnehc/OpenVAF

another one ?
have a documentation where we can know whats the specificity of all these fork ?
which one we definitly need to use ?

jalcim, 2025-03-08:
and what is the difference betwin the `verilogAE` tag and other ?

Kreijstal, 2025-03-08:
just use https://github.com/arpadbuermen/OpenVAF, no?

jalcim, 2025-03-09:
and why not https://github.com/OpenVAF/OpenVAF-Reloaded so ?

jalcim, 2025-03-09:
> just use https://github.com/arpadbuermen/OpenVAF, no?

because this repository is not maintained since 2 years

jalcim, 2025-03-09:
i'm totaly lost between all these fork and version xD

arpadbuermen, 2025-05-17:
https://github.com/OpenVAF/OpenVAF-Reloaded is maintained by @Kreijstal. It mostly follows my repo at  https://github.com/arpadbuermen/OpenVAF and has support for llvm18. 

My repo is probably the bleeding edge regarding features and fixes with the exception of llvm18 support. Yesterday I did some benchmarking of llvm 15 vs 16 and there was no significant difference. Don't know about 18, though. 

This repo is unfortunately dead afaik. 

@Kreijstal ... there are some changes you can merge from my repo...

Kreijstal, 2025-05-17:
> https://github.com/OpenVAF/OpenVAF-Reloaded is maintained by [@Kreijstal](https://github.com/Kreijstal). It mostly follows my repo at https://github.com/arpadbuermen/OpenVAF and has support for [...]

will do, unfortunately tight this semester. I'll find the time