pascalkuthe/OpenVAF #156: ddt() causes singular matrix

serjaio, 2025-03-12:
Casting the derivative of a voltage to a voltage brings about a "singular matrix" in NGSPICE and fails to converge.
Here's a test file debug.va

```
`include "discipline.h"
module modulename (x,debug);
  inout 		x,debug;
  electrical 	x,debug;
  parameter real alpha=0.2;

  analog begin
	//I(x)	    <+ ddt( V(x) );
	V(debug)	<+ ddt( V(x) );
  end
endmodule

```

I simulated with the file bench.sp:

```
V0 x 0 SIN 0 20 100 0 20
NDUT x debug coco
.model coco modulename
.control 
pre_osdi debug.osdi
tran 1e-6 100e-3
plot v(x) 
plot i(V0) 
nodeset all=0;
.endc
.end
```

On the other hand, the commented line, casting the derivative of a voltage to a current, behaving like a capacitor,  works fine.
I am using ngspice-44 and openvaf 23.5.0.

serjaio, 2025-03-12:
The messages of NGSPICE:
```
******
** ngspice-44 : Circuit level simulation program
** Compiled with KLU Direct Linear Solver
** The U. C. Berkeley CAD Group
** Copyright 1985-1994, Regents of the University of California.
** Copyright 2001-2024, The ngspice team.
** Please get your ngspice manual from https://ngspice.sourceforge.io/docs.html
** Please file your bug-reports at https://ngspice.sourceforge.io/bugrep.html
** Creation Date: Dec 29 2024   15:21:16
******

Note: No compatibility mode selected!

Circuit: 

Doing analysis at TEMP = 27.000000 and TNOM = 27.000000

Using SPARSE 1.3 as Direct Linear Solver
Warning: singular matrix:  check nodes ndut#flow(debug) and debug

Note: Starting dynamic gmin stepping
Warning: singular matrix:  check nodes ndut#flow(debug) and debug

Warning: singular matrix:  check nodes ndut#flow(debug) and debug

Warning: singular matrix:  check nodes ndut#flow(debug) and debug

Warning: singular matrix:  check nodes ndut#flow(debug) and debug

Warning: singular matrix:  check nodes ndut#flow(debug) and debug

Warning: Dynamic gmin stepping failed
Note: Starting true gmin stepping
Warning: True gmin stepping failed
Note: Starting source stepping
Warning: source stepping failed
Note: Transient op started
Error: Transient op failed, timestep too small

Error: The operating point could not be simulated successfully.
    Any of the following steps may fail.!

Transient solution failed -

Last Node Voltages
------------------

Node                                   Last Voltage        Previous Iter
----                                   ------------        -------------
x                                                 0                    0
debug                                             0                    0
v0#branch                                         0                    0

No. of Data Rows : 0
doAnalyses: TRAN:  Timestep too small; initial timepoint: cause unrecorded.

tran simulation(s) aborted
Error: no such vector v(x)
Error: no such vector i(v0)
nodeset: no such command available in ngspice
ngspice 1 -> 
Interrupted once . . .
Warning: clearing control structures
ngspice 1 -> 
```

serjaio, 2025-03-12:
Sorry Kreijstal. 
I don't know what your are referring to.

dwarning, 2025-03-12:
Spice files need a title line - and everything is well.