pascalkuthe/OpenVAF #157: Openvaf-compiled model behaivor

modestpe, 2025-04-05:
I have a simple memory device verilog-a model. Was able to compile and start to simulate. 

variable "statei"  was used to represent programmed state (= 1) or erased state (= 0) of the memory device.

When the memory device is in the programmed state, the resistance of the device becomes lower compared to the erased state. 

The variable "statei" can transit to the target state with a programming pulse or an erasing pulse.

However, the variable "statei" reverts as soon as the programming pulse or erasing pulse is removed. 

I have attached captured images and source codes to reproduce the error. 

when we simulate the same code using Cadence verilog compiler and simulator. didn't observe this issue. 

[z2tps.zip](https://github.com/user-attachments/files/19611375/z2tps.zip)

Kreijstal, 2025-04-05:
This repo isn't mantained

Lang0722, 2025-04-07:
Maybe a Hidden states issue.

felix-salfelder, 2025-04-09:
The replies so far may be well intentioned, but fall short of an explanation what's going on.

OpenVAF works well with pure-analog aka. Spice compact device models, subject to the usual issues with Spice itself. The development here has basically stopped, but a few copies are maintained (elsewhere). Perhaps Kreijstal suggests to use a copy he is involved with.

The term "hidden state" was introduced to characterise requirements of closed source proprietary implementations. It is not clear how it can be used to describe the mishap in OpenVAF. OpenVAF is completely "open" as the name suggests and apparently states are not implemented at all. It is pretty clear what the standard requires, so this would need fixing.

Beyond states, hidden or not, in order to describe and simulate memory models (correctly), Verilog-A has significantly more on offer. Sensitivity to region changes should be modelled using cross events and output should be filtered, as the commented out "transition" line suggests. See LRM Section 5.10.3.1 for an explanation and for sample-and-hold examples much simpler than the memory model shown above.

kev-cam, 2025-04-09:
For anyone interested in low-power digital design (DVFS), here's a modeling
technique that works in Verilog-A -

http://www.v-ms.com/ICCAD-2014.pdf

Regards,
Kev.

arpadbuermen, 2025-05-17:
Afaik Verilog-A variables retain their value between model evaluations. Such a variable is referred to as a hidden state. The simplest way to prevent a variable from introducing a hidden state is to initialize a it. 
Compact models of transistors are not supposed to rely on hidden states because if they do, then it is hard to formulate advanced analyses, like HB, since the model's response cannot be expressed as a solution of a PDE with excitation (model input) on its RHS. 
Commercial simulators detect hidden state and prevent simulation if such a model is used. There's a discussion on this subject somewhere on Cadence forums. Hidden state its not some commercial bruhaha. Because FOSS simulators do only basic SPICE analyses this is not an issue in the FOSS world. 
OpenVAF does not support hidden state. A variable's value is not retained between model invocations. Judging from comments Pascal had plans to do this in the future. It is possible and would involve allocating a state in the simulator's state vector where the variable would be stored.