pascalkuthe/OpenVAF #159: Verilog-A slew implementation

holmesozic, 2025-05-07:
Please consider this request for implementing slew in OpenVAF. 

How can I help?

Kreijstal, 2025-05-07:
ask on arpad´s fork

felix-salfelder, 2025-05-15:
> ask on arpad`s fork

@JavaNoviceProgrammer asked a very similar question there, see https://github.com/arpadbuermen/OpenVAF/issues/12.

Sadly, Arpad has already closed the issue and deleted my comment there. Maybe he is short on time. Anyway, I like dispute more than censorship.

My comment basically said that implementing an OSDI wrapper for Gnucap is a good start. Learn what's missing in the interface, compare with a complete implementation. Then try and upgrade OSDI, maybe bump OpenVAF.

We at Gnucap have NLnet based funding for guided internship work on Verilog-AMS, standardisation and related topics.